UZ-SLAMLab/ORB_SLAM3 #478: Airsim ROS with ORB SLAM3

yukaryote, 2022-01-21:
I'm trying to test ORB SLAM3's Monocular ROS node on AirSim. I'm subscribing to AirSim's camera topic by changing line 62 of https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/Examples_old/ROS/ORB_SLAM3/src/ros_mono.cc to the AirSim camera topic and the topic name is correct. However, I see a black screen and a "waiting for images" message from ORB SLAM's GUI. Also, the same issue arises when I try to visualize a rosbag from the EuRoC dataset.

I have a feeling this may have happened because I had to follow the instructions in #442 in order to actually build the ROS package. Specifically, I had a error that was something like`invalid conversion between Sophus::SE3f, cv::MAT,Eigen::Vector3f in cv::Mat Tcw = mpSLAM->TrackMonocular(cv_ptr->image,cv_ptr->header.stamp.toSec());` in the `ros_mono_ar.cc` file.

closed 2022-01-21 by yukaryote

Maithilishetty, 2022-03-02:
@yukaryote I have the same issue while running the rosbag from the EuROC dataset. Could you please tell me how to fix this? Thank you!

yukaryote, 2022-03-03:
@Maithilishetty I fixed this by using a docker container for ORB SLAM 3 instead. This one works for me: https://github.com/jahaniam/orbslam3_docker

KhazixW2, 2023-04-14:
hi，i have solve the problem , because orbslam3 have a map from /camere/image_view  to /cam/image_view ，the "/cam0/image_view" is you camera name. 

have a example ,  i have a camera "camera_ssz1 "  than  the /cam/image_view should be update /camera_ssz1/image_view

the update you have to set , in  launch dirctory .