UZ-SLAMLab/ORB_SLAM3 #479: 'build_ros.sh' is waiting for updating

Yi-Zhou97, 2022-01-21:
When building the ROS version, there are some problems, such as the confusion between the file 'Examples_old' and 'Examples'. 

Also, the process of 'make -j' will bring some problems as following (one as e.g.):

> error: conversion from ‘Eigen::Vector3f’ {aka ‘Eigen::Matrix<float, 3, 1>’} to non-scalar type ‘cv::Mat’ requested
>   530 |             cv::Mat Xw = pMP->GetWorldPos();

JACKLiuDay, 2022-01-24:
https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/442 
Hi, you can try this.
Or, if you don't need the AR function, just delete it in the Cmakelists.

Qiu1117, 2022-01-27:
> #442 Hi, you can try this. Or, if you don't need the AR function, just delete it in the Cmakelists.

After I used the modification of point 4 in this link to replace the error of the original code, I got some new errors.

> /home/qiuf/catkin_ws/src/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/src/AR/ViewerAR.cc:406:36: error: invalid initialization of reference of type ‘cv::InputArray’ {aka ‘const cv::_InputArray&’} from expression of type ‘Eigen::Vector3f’ {aka ‘Eigen::Matrix<float, 3, 1>’}
  406 |          cv::eigen(pMP->GetWorldPos(), WorldPos);
      |                    ~~~~~~~~~~~~~~~~^~

> /home/qiuf/catkin_ws/src/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/src/AR/ViewerAR.cc:533:33: error: invalid initialization of reference of type ‘cv::InputArray’ {aka ‘const cv::_InputArray&’} from expression of type ‘Eigen::Vector3f’ {aka ‘Eigen::Matrix<float, 3, 1>’}
  533 |       cv::eigen(pMP->GetWorldPos(), Xw);
      |                 ~~~~~~~~~~~~~~~~^~

> /home/qiuf/catkin_ws/src/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/src/AR/ros_mono_ar.cc:154:15: error: invalid initialization of reference of type ‘cv::InputArray’ {aka ‘const cv::_InputArray&’} from expression of type ‘Eigen::Matrix4f’ {aka ‘Eigen::Matrix<float, 4, 4>’}
  154 |     cv::eigen(Tcw_Matrix, Tcw);
      |               ^~~~~~~~~~

JACKLiuDay, 2022-01-28:
@Qiu1117 Hi, I just see your reply.  The errors seem to locate at the AR function. I just delete the related add_executable in the CMakeLists as I don't use the AR function. Maybe you can try this.

Qiu1117, 2022-01-28:
> @Qiu1117 Hi, I just see your reply. The errors seem to locate at the AR function. I just delete the related add_executable in the CMakeLists as I don't use the AR function. Maybe you can try this.

It works！Thank you. : )

anchao1997, 2022-03-12:
### in AR of Examples_old
# 

```c++
#include "../../../include/Converter.h"
```

error: conversion from ‘Sophus::SE3f’ {aka ‘Sophus::SE3<float>’} to non-scalar type ‘cv::Mat’ requested
  151 |     cv::Mat Tcw = mpSLAM->TrackMonocular(cv_ptr->image,cv_ptr->header.stamp.toSec());

```c++
cv::Mat Tcw=ORB_SLAM3::Converter::toCvMat(mpSLAM->TrackMonocular(cv_ptr->image,cv_ptr->header.stamp.toSec()).matrix());
//in line 151 of ros_mono_ar.cc
```

error: no matching function for call to ‘std::vector<cv::Mat>::push_back(Eigen::Vector3f)’
  405 |                 vPoints.push_back(pMP->GetWorldPos());

```c++
vPoints.push_back(ORB_SLAM3::Converter::toCvMat(pMP->GetWorldPos()));
//in line 405 of ViewerAR.cc
```

error: conversion from ‘Eigen::Vector3f’ {aka ‘Eigen::Matrix<float, 3, 1>’} to non-scalar type ‘cv::Mat’ requested
  530 |             cv::Mat Xw = pMP->GetWorldPos();

```c++
cv::Mat Xw = ORB_SLAM3::Converter::toCvMat(pMP->GetWorldPos());
//in line 530 of ViewerAR.cc
```

all above .cc file should include Converter.h file

NguyenCanhThanh, 2022-10-28:
> ### in AR of Examples_old
> ```c++
> #include "../../../include/Converter.h"
> ```
> 
> error: conversion from ‘Sophus::SE3f’ {aka ‘Sophus::SE3’} to non-scalar type ‘cv::Mat’ requested 151 | cv::Mat Tcw = mpSLAM->TrackMonocular(cv_ptr->image,cv_ptr->header.stamp.toSec());
> 
> ```c++
> cv::Mat Tcw=ORB_SLAM3::Converter::toCvMat(mpSLAM->TrackMonocular(cv_ptr->image,cv_ptr->header.stamp.toSec()).matrix());
> //in line 151 of ros_mono_ar.cc
> ```
> 
> error: no matching function for call to ‘std::vectorcv::Mat::push_back(Eigen::Vector3f)’ 405 | vPoints.push_back(pMP->GetWorldPos());
> 
> ```c++
> vPoints.push_back(ORB_SLAM3::Converter::toCvMat(pMP->GetWorldPos()));
> //in line 405 of ViewerAR.cc
> ```
> 
> error: conversion from ‘Eigen::Vector3f’ {aka ‘Eigen::Matrix<float, 3, 1>’} to non-scalar type ‘cv::Mat’ requested 530 | cv::Mat Xw = pMP->GetWorldPos();
> 
> ```c++
> cv::Mat Xw = ORB_SLAM3::Converter::toCvMat(pMP->GetWorldPos());
> //in line 530 of ViewerAR.cc
> ```
> 
> all above .cc file should include Converter.h file

Tkanks bro ^^

thehuongrbe, 2022-11-11:
> Tkanks bro ^^

Hi bro. i had the same error when try build it on ros2. How can fix it? Hope to see you soon! <3
My log error:
 error: conversion from ‘Sophus::SE3f’ {aka ‘Sophus::SE3<float>’} to non-scalar type ‘cv::Mat’ requested
   52 |     cv::Mat Tcw = m_SLAM->TrackRGBD(cv_ptrRGB->image, cv_ptrD->image, msgRGB->header.stamp.sec);

anchao1997, 2022-11-11:
> > > ### in AR of Examples_old
> > > ```c++
> > > #include "../../../include/Converter.h"
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > error: conversion from ‘Sophus::SE3f’ {aka ‘Sophus::SE3’} to non-scalar type ‘cv::Mat’ requested 151 | cv::Mat Tcw = mpSLAM->TrackMonocular(cv_ptr->image,cv_ptr->header.stamp.toSec());
> > > ```c++
> > > cv::Mat Tcw=ORB_SLAM3::Converter::toCvMat(mpSLAM->TrackMonocular(cv_ptr->image,cv_ptr->header.stamp.toSec()).matrix());
> > > //in line 151 of ros_mono_ar.cc
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > error: no matching function for call to ‘std::vectorcv::Mat::push_back(Eigen::Vector3f)’ 405 | vPoints.push_back(pMP->GetWorldPos());
> > > ```c++
> > > vPoints.push_back(ORB_SLAM3::Converter::toCvMat(pMP->GetWorldPos()));
> > > //in line 405 of ViewerAR.cc
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > error: conversion from ‘Eigen::Vector3f’ {aka ‘Eigen::Matrix<float, 3, 1>’} to non-scalar type ‘cv::Mat’ requested 530 | cv::Mat Xw = pMP->GetWorldPos();
> > > ```c++
> > > cv::Mat Xw = ORB_SLAM3::Converter::toCvMat(pMP->GetWorldPos());
> > > //in line 530 of ViewerAR.cc
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > all above .cc file should include Converter.h file
> > 
> > 
> > Tkanks bro ^^
> 
> Hi bro. i had the same error when try build it on ros2. How can fix it? Hope to see you soon! <3 My log error: error: conversion from ‘Sophus::SE3f’ {aka ‘Sophus::SE3’} to non-scalar type ‘cv::Mat’ requested 52 | cv::Mat Tcw = m_SLAM->TrackRGBD(cv_ptrRGB->image, cv_ptrD->image, msgRGB->header.stamp.sec);

I think you use ORB_SLAM3::Converter::toCvMat function to do so.

cwg968, 2023-06-21:
error: conversion from ‘Sophus::SE3f’ {aka ‘Sophus::SE3’} to non-scalar type ‘cv::Mat’ requested 151 | cv::Mat Tcw = mpSLAM->TrackMonocular(cv_ptr->image,cv_ptr->header.stamp.toSec());
`
cv::Mat Tcw;
Sophus::SE3f Tcw_SE3f = mpSLAM->TrackMonocular(cv_ptr->image,cv_ptr->header.stamp.toSec());
Eigen::Matrix4f Tcw_Matrix = Tcw_SE3f.matrix();
cv::eigen2cv(Tcw_Matrix, Tcw);
// in line 151 of ros_mono_ar.cc
`
error: no matching function for call to ‘std::vectorcv::Mat::push_back(Eigen::Vector3f)’ 405 | vPoints.push_back(pMP->GetWorldPos());

`
cv::Mat WorldPos;
cv::eigen2cv(pMP->GetWorldPos(), WorldPos);
vPoints.push_back(WorldPos);
//in line 405 of ViewerAR.cc
`
error: conversion from ‘Eigen::Vector3f’ {aka ‘Eigen::Matrix<float, 3, 1>’} to non-scalar type ‘cv::Mat’ requested 530 | cv::Mat Xw = pMP->GetWorldPos();

`
cv::Mat Xw;
cv::eigen2cv(pMP->GetWorldPos(), Xw);
//in line 530 of ViewerAR.cc
`
and you should correct the code before add
`
#include <Eigen/Dense>
#include<opencv2/core/eigen.hpp>
`
all above .cc file should include add this!

joooohnson, 2023-07-08:
> ### in AR of Examples_old
>
> ```c++
> #include "../../../include/Converter.h"
> ```
>
> error: conversion from ‘Sophus::SE3f’ {aka ‘Sophus::SE3’} to non-scalar type ‘cv::Mat’ requested 151 | cv::Mat Tcw = [...]

Thank u <3