UZ-SLAMLab/ORB_SLAM3 #483: ROS build fails

smileyenot983, 2022-02-01:
Hi everyone, i cannot manage with building ORB slam 3 for ROS. Building orb slam 3 was successful

I have following errors:

/home/ramil/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/src/AR/ViewerAR.cc:405:53: error: no matching function for call to ‘std::vector<cv::Mat>::push_back(Eigen::Vector3f)’
                 vPoints.push_back(pMP->GetWorldPos());

/usr/include/c++/7/bits/stl_vector.h:953:7: note:   no known conversion for argument 1 from ‘Eigen::Vector3f {aka Eigen::Matrix<float, 3, 1>}’ to ‘std::vector<cv::Mat>::value_type&& {aka cv::Mat&&}’
/home/ramil/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/src/AR/ViewerAR.cc: In member function ‘void ORB_SLAM3::Plane::Recompute()’:
/home/ramil/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/src/AR/ViewerAR.cc:530:42: error: conversion from ‘Eigen::Vector3f {aka Eigen::Matrix<float, 3, 1>}’ to non-scalar type ‘cv::Mat’ requested
             cv::Mat Xw = pMP->GetWorldPos();

/home/ramil/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/src/AR/ros_mono_ar.cc: In member function ‘void ImageGrabber::GrabImage(const ImageConstPtr&)’:
/home/ramil/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/src/AR/ros_mono_ar.cc:151:41: error: conversion from ‘Sophus::SE3f {aka Sophus::SE3<float>}’ to non-scalar type ‘cv::Mat’ requested
     cv::Mat Tcw = mpSLAM->TrackMonocular(cv_ptr->image,cv_ptr->header.stamp.toSec());

And at the end i have this:
[ 92%] Linking CXX executable ../Stereo
/usr/bin/ld: warning: libopencv_highgui.so.405, needed by ../../../../lib/libORB_SLAM3.so, may conflict with libopencv_highgui.so.3.2
/usr/bin/ld: warning: libopencv_features2d.so.405, needed by ../../../../lib/libORB_SLAM3.so, may conflict with libopencv_features2d.so.3.2
/usr/bin/ld: warning: libopencv_imgproc.so.405, needed by ../../../../lib/libORB_SLAM3.so, may conflict with libopencv_imgproc.so.3.2
/usr/bin/ld: warning: libopencv_core.so.405, needed by ../../../../lib/libORB_SLAM3.so, may conflict with libopencv_core.so.3.2
[ 92%] Built target Stereo
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

nindanaoto, 2022-02-21:
This is the fixed fork.
https://github.com/nindanaoto/ORB_SLAM3

robin-shaun, 2022-03-13:
@nindanaoto Hello, which eigen version is in your computer? My eigen version is 3.4.0, and I met a series of errors like this
`/usr/include/eigen3/unsupported/Eigen/CXX11/src/util/EmulateArray.h:258:30: error: redeclared with 1 template parameter  template <typename T> struct array_size;`

AiYoWeiYL, 2022-09-04:
> > @nindanaoto Hello, which eigen version is in your computer? My eigen version is 3.4.0, and I met a series of errors like this `/usr/include/eigen3/unsupported/Eigen/CXX11/src/util/EmulateArray.h:258:30: error: redeclared with 1 template parameter template <typename T> struct array_size;`
> 
> I met the same question,my eigen version is 3.2.1, have you fix it?

Hi, have you solved this problem？ I met the same problem

nindanaoto, 2022-09-04:
Oh, I did not notice there are a lot of people encountering same problem. I'm using 3.3.7-2, default version of official Ubuntu 20.04's repository.
I wrote the instruction to use ORB SLAM3 with OAK-D on LXD. This may help you to solve this issue.
https://qiita.com/nindanaoto/items/20858eca08aad90b5bab

AiYoWeiYL, 2022-09-05:
I'll give it a try. Thank you

kenhuang1964, 2023-11-21:
Hey @nindanaoto, do you know if ORB-SLAM3 works with OAK-D Pro?

nindanaoto, 2023-12-31:
Hi, I already noticed this message and still couldn't find the time to reproduce the result in the current environments. As far as I remember, ORB-SLAM3 can work with OKA-D Pro for stereo settings but not RGBD. I hope to retry the RGBD setting with the current software environments.....

kenhuang1964, 2023-12-31:
> Hi, I already noticed this message and still couldn't find the time to reproduce the result in the current environments. As far as I remember, ORB-SLAM3 can work with OKA-D Pro for stereo settings but not RGBD. I hope to retry the RGBD setting with the current software environments.....

Cool, thanks!

sirackerman, 2024-08-30:
> This is the fixed fork. https://github.com/nindanaoto/ORB_SLAM3

worked liked a charm. Thank you

parastoobakhtiarii, 2025-02-06:
@nindanaoto 

Hi, have you tried using ros2, oak-d pro and ORB-SLAM3?