UZ-SLAMLab/ORB_SLAM3 #486: Disabling GUI display

MarcusDudeBro, 2022-02-09:
I'm trying to run ORB SLAM 3 on a Jetson Xavier NX board with ubuntu, however, I don't have a GUI on it.  Is there any way I can disable the pangolin implemented GUI display for ORB SLAM? Thanks.

closed 2022-02-09 by MarcusDudeBro

man-do, 2022-02-14:
How did you disable the display ?

MarcusDudeBro, 2022-02-14:
> How did you disable the display ?

The 4th parameter for the main SLAM system Initializer System() is a boolean value responsible for enabling the GUI (true is enabled, false is disabled)

ergys00, 2023-02-02:
Hi there, when i try to launch Pangolin i get that error 
error: XDG_RUNTIME_DIR not set in the environment. terminate called after throwing an instance of 'std::runtime_error'   what():  Pangolin X11: Failed to open X display Aborted (core dumped)

I've tried chaning the bUseViewer variable to false and recompiling and installing but it doesn't work and still gives that error, do you have any advice?