UZ-SLAMLab/ORB_SLAM3 #487: Why is the accuracy getting worse ?

chennuo0125-HIT, 2022-02-09:
[image]

from above picture, in stereo mode, ORB_SLAM2 get better accuracy than ORB_SLAM3, What changes in ORB_SLAM3 lead to this result ?

You-zeyi, 2022-02-10:
> [image]
> 
> from above picture captured from paper, in stereo mode, ORB_SLAM2 get better accuracy than ORB_SLAM3, What changes in ORB_SLAM3 lead to this result ?

You can clearly see from the table that ORB_SLAM2 failed to finish the last data sequence but ORB_SLAM3 succeeded with a low accuracy. So the accuracy of ORB_SLAM2 does not include the result from last sequence. that is why its accuracy are marked with a * sign.

chennuo0125-HIT, 2022-02-11:
> You can clearly see from the table that ORB_SLAM2 failed to finish the last data sequence but ORB_SLAM3 succeeded with a low accuracy. So the accuracy of ORB_SLAM2 does not include the result from [...]

closed 2022-02-11 by chennuo0125-HIT