UZ-SLAMLab/ORB_SLAM3 #489: "nan" (not a number) average, std time duration values when REGISTER_TIMES set

brianm-sra, 2022-02-09:
I uncommented this line in include/Settings.h :

#define REGISTER_TIMES

This is the "Flag to activate the measurement of time in each process (track,localmap, place recognition)."

I then ran some EuRoC and TUM-VI data sets 

The output for each included a number of "nan" (not a number) results, such as shown in the quoted block below.

Has anyone else seen this?   Any idea what the root cause could be?

```
Loop Closing (mean$\pm$std)
Loop Fusion: -nan$\pm$-nan
Essential Graph: -nan$\pm$-nan
Total Loop Closing: -nan$\pm$-nan
Num exec: 0
Number of KFs: -nan$\pm$-nan
Map Merging (mean$\pm$std)
Merge Maps: -nan$\pm$-nan
Welding BA: -nan$\pm$-nan
Optimization Ess.: -nan$\pm$-nan
Total Map Merging: -nan$\pm$-nan
Num exec: 0
Number of KFs: -nan$\pm$-nan
Number of MPs: -nan$\pm$-nan
Full GBA (mean$\pm$std)
GBA: -nan$\pm$-nan
Map Update: -nan$\pm$-nan
Total Full GBA: -nan$\pm$-nan
Num exec: 0
Num abort: 0
Number of KFs: -nan$\pm$-nan
Number of MPs: -nan$\pm$-nan
```

brianm-sra, 2022-02-11:
I do get some additional numeric values for EuRoC V102 stereo only and V103 stereo only
 (such as Loop Fusion, Total Loop Closing, GBA, Total Full GBA) 

So it appears to be an issue only for some datasets, not all

brianm-sra, 2022-02-11:
I still saw nan for Map Merging.  Under what circumstances does Map Merging happen?

mel-ctrl, 2022-04-12:
When I add #define REGISTER_TIMES in the system header file I get a segmentation fault after the viewer starts. Did you also encounter this problem @brianm-sra ?

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Stereo
Loading settings from ./Examples/Stereo/Rosario2.yaml
	-Loaded camera 1
	-Loaded camera 2
Camera.newHeight optional parameter does not exist...
Camera.newWidth optional parameter does not exist...
	-Loaded image info
	-Loaded ORB settings
Viewer.imageViewScale optional parameter does not exist...
	-Loaded viewer settings
System.LoadAtlasFromFile optional parameter does not exist...
System.SaveAtlasToFile optional parameter does not exist...
	-Loaded Atlas settings
System.thFarPoints optional parameter does not exist...
	-Loaded misc parameters
----------------------------------
SLAM settings: 
	-Camera 1 parameters (Pinhole): [ 348.522 348.45 344.484 188.808 ]
	-Camera 2 parameters (Pinhole	-Original image size: [ 672 , 376 ]
	-Current image size: [ 672 , 376 ]
	-Sequence FPS: 15
	-Stereo baseline: 0.118718
	-Stereo depth threshold : 41.2728
	-Features per image: 1250
	-ORB scale factor: 1.2
	-ORB number of scales: 8
	-Initial FAST threshold: 20
	-Min FAST threshold: 7

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
There are 1 cameras in the atlas
Camera 0 is pinhole
Starting the Viewer
Segmentation fault (core dumped)