UZ-SLAMLab/ORB_SLAM3 #491: ROS Build Error: Unable to Locate Sophus Thirdparty Files

CWAndersn, 2022-02-10:
I have encountered the following error when attempting to run the build_ros.sh script. It seems there is something wrong with the build process that prevents the Sophus library froum being found. Any idea why this is?

```
In file included from /ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/../../../include/Frame.h:30:0,
                 from /ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/../../../include/KeyFrame.h:28,
                 from /ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/../../../include/MapPoint.h:23,
                 from /ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/../../../include/FrameDrawer.h:24,
                 from /ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/../../../include/Viewer.h:23,
                 from /ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/../../../include/Tracking.h:26,
                 from /ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/../../../include/System.h:31,
                 from /ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/src/AR/ViewerAR.h:27,
                 from /ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/src/AR/ViewerAR.cc:19:
/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/../../../include/ImuTypes.h:29:10: fatal error: sophus/se3.hpp: No such file or directory
 #include <sophus/se3.hpp>
```

man-do, 2022-02-14:
Hi,
I think you have to build the manually, just go on Thirdparty folder and build the libs manually.

ACBBZ, 2022-02-20:
I have the same problem. I want to know how to deal with this problem. The library of Thirdparty folder has been built, but the ros_build.sh can not find the library.

nindanaoto, 2022-02-21:
You have to add it to CMakeList.txt like this.
https://github.com/nindanaoto/ORB_SLAM3/blob/ec9ea0a24b4c5e2181a912751ad01bd17d31ea46/Examples/ROS/ORB_SLAM3/CMakeLists.txt#L46

mahammadirfan, 2022-02-23:
I've met the same problem while installing the latest repository. The installation script build.sh tells you that running **cmake ..**  and then **make** in sophus build directory is enough but you need to run **sudo make install** as well and the problem sophus/se3.hpp will be resolved.

tan-may16, 2022-04-23:
Hi,

I am facing the same exact issue and running sudo make install did not resolve it. @nindanaoto I already had that in the CMakeList.txt file. I am copying the error for your reference below.

/ImuTypes.h:29:10: fatal error: sophus/se3.hpp: No such file or directory
 #include <sophus/se3.hpp>
          ^~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/MonoAR.dir/build.make:198: recipe for target 'CMakeFiles/MonoAR.dir/src/AR/ViewerAR.cc.o' failed
make[2]: *** [CMakeFiles/MonoAR.dir/src/AR/ViewerAR.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....

and then it is repeated for 2-3 times.

Thank you

nindanaoto, 2022-09-04:
Here is the instruction for ORB-SLAM3 installing with modified my ORB SLAM3 fork. I guess it may help to solve this issue.
https://qiita.com/nindanaoto/items/20858eca08aad90b5bab

virajawate, 2023-09-08:
I installed Sophus separatly

hussains72, 2024-12-14:
If sophus is installed and you cant find it.
Add this in CMakeLists.txt
You need to add the path to the sophus folder. Update the include_directories section as follows:

${PROJECT_SOURCE_DIR}/../../../Thirdparty/Sophus