UZ-SLAMLab/ORB_SLAM3 #493: Strange behavior trying to load SLAM settings

jamestkpoon, 2022-02-14:
Hi! I have the following rather basic settings yaml for a ZED2 camera:

```
%YAML:1.0

#--------------------------------------------------------------------------------------------
# System config
#--------------------------------------------------------------------------------------------

# When the variables are commented, the system doesn't load a previous session or not store the current one

# If the LoadFile doesn't exist, the system give a message and create a new Atlas from scratch
#System.LoadAtlasFromFile: "Session_MH01_MH02_MH03_Stereo60_Pseudo"

# The store file is created from the current session, if a file with the same name exists it is deleted
#System.SaveAtlasToFile: "Session_MH01_MH02_MH03_Stereo60_Pseudo"

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------
File.version: "1.0"

Camera.type: "Rectified"

# Camera calibration and distortion parameters (OpenCV) 
Camera1.fx: 522.6021728515625
Camera1.fy: 522.6021728515625
Camera1.cx: 650.2871704101562
Camera1.cy: 359.57666015625

Camera.width: 1280
Camera.height: 720

# Camera frames per second 
Camera.fps: 30

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 0

Stereo.ThDepth: 60.0
Stereo.b: 0.12

#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 1200

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.2

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 8

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#--------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1.0
Viewer.GraphLineWidth: 0.9
Viewer.PointSize: 2.0
Viewer.CameraSize: 0.08
[... 6 lines cut ...]
```

but when I try to create a new ORB_SLAM3::System I get something like:
```
ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Stereo
Loading settings from /ros2_ws/src/ros2-orbslamv3-wrapper/config/zed2.yaml
	-Loaded camera 1
	-Loaded camera 2
	-Loaded image info
	-Loaded ORB settings
	-Loaded viewer settings
	-Loaded Atlas settings
	-Loaded misc parameters
----------------------------------
SLAM settings: 
	-Camera 1 parameters (Pinhole): [ 522.602 522.602 650.287 359.577 ]
	-Camera 2 parameters (Pinhole: [ 7.39886e-43 0 7.39886e-43 0 9.40395e-38 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 -2 0.00195313 1.77965e-43 0 -0 0.00195312 -1.77965e-43 0.00195312 -2.52439e-29 0.00195312 -2.52439e-29 0.00195312 -1.08422e-19 0.00195312 -1.08422e-19 0.00195312 -4.65668e-10 0.00195312 -4.65668e-10 0.00195312 -2.00003 0.00195312 -2.00003 0.00195312 -8.59006e+09 0.00195312 -8.59006e+09 0.00195312 -3.6894e+19 0.00195312 -3.6894e+19 0.00195312 -1.58459e+29 0.00195312 -1.58459e+29 0.00195312 1.77965e-43 0.00195313 1.77965e-43 0.00195313 2.52439e-29 0.00195313 2.52439e-29 0.00195313 1.08422e-19 0.00195313 1.08422e-19 0.00195313 4.65668e-10 0.00195313 4.65668e-10 0.00195313 2.00003 0.00195313 2.00003 0.00195313 8.59006e+09 0.00195313 8.59006e+09 0.00195313 3.6894e+19 0.00195313 3.6894e+19 0.00195313 1.58459e+29 0.00195313 1.58459e+29 0.00195313 -1.77965e-43 0.00195313 -1.77965e-43 0.00195313 -2.52439e-29 0.00195313 -2.52439e-29 0.00195313 -1.08422e-19 0.00195313 -1.08422e-19 0.00195313 -4.65668e-10 0.00195313 -4.65668e-10 0.00195313 -2.00003 0.00195313 -2.00003 0.00195313 -8.59006e+09 0.00195313 -8.59006e+09 0.00195313 -3.6894e+19 0.00195313 -3.6894e+19 0.00195313 -1.58459e+29 0.00195313 -1.58459e+29 0.00195313 1.77965e-43 0.00195313 1.77965e-43 0.00195313 2.52439e-29 0.00195313 2.52439e-29 0.00195313 1.08422e-19 0.00195313 1.08422e-19 0.00195313 4.65668e-10 0.00195313 4.65668e-10 0.00195313 2.00003 0.00195313 2.00003 0.00195313 8.59006e+09 0.00195313 8.59006e+09 0.00195313 3.6894e+19 0.00195313 3.6894e+19 0.00195313 1.58459e+29 0.00195313 1.58459e+29 0.00195313 -1.77965e-43 0.00195313 -1.77965e-43 0.00195313 -2.52439e-29 0.00195313 -2.52439e-29 0.00195313 -1.08422e-19 0.00195313 -1.08422e-19 0.00195313 -4.65668e-10 0.00195313 -4.65668e-10 0.00195313 -2.00003 0.00195313 -2.00003 0.00195313 -8.59006e+09 0.00195313 -8.59006e+09 0.00195313 -3.6894e+19 0.00195313 -3.6894e+19 0.00195313 -1.58459e+29 0.00195313 -1.58459e+29 0.00195313 1.77965e-43 0.00195313 1.77965e-43 0.00195313 2.52439e-29 0.00195313 2.52439e-29 0.00195313 1.08422e-19 0.00195313 1.08422e-19 0.00195313 4.65668e-10 0.00195313 4.65668e-10 0.00195313 2.00003 0.00195313 2.00003 0.00195313 8.59006e+09 0.00195313 8.59006e+09 0.00195313 3.6894e+19 0.00195313 3.6894e+19 0.00195313 1.58459e+29 0.00195313 1.58459e+29 0.00195313 -1.77965e-43 0.00195313 -1.77965e-43 0.00195313 -2.52439e-29 0.00195313 -2.52439e-29 0.00195313 -1.08422e-19 0.00195313 -1.08422e-19 0.00195313 -4.65668e-10 0.00195313 -4.65668e-10 0.00195313 -2.00003 0.00195313 -2.00003 0.00195313 -8.59006e+09 0.00195313 -8.59006e+09 0.00195313 -3.6894e+19 0.00195313 -3.6894e+19 0.00195313 -1.58459e+29 0.00195313 -1.58459e+29 0.00195313 1.77965e-43 0.00195313 1.77965e-43 0.00195313 2.52439e-29 0.00195313 2.52439e-29 0.00195313 1.08422e-19 0.00195313 1.08422e-19 0.00195313 4.65668e-10 0.00195313 4.65668e-10 0.00195313 2.00003 0.00195313 2.00003 0.00195313 8.59006e+09 0.00195313 8.59006e+09 0.00195313 3.6894e+19 0.00195313 3.6894e+19 0.00195313 1.58459e+29 0.00195313 1.58459e+29 0.00195313 -1.77965e-43 0.00195313 -1.77965e-43 0.00195313 -2.52439e-29 0.00195313 -2.52439e-29 0.00195313 -1.08422e-19 0.00195313 -1.08422e-19 0.00195313 -4.65668e-10 0.00195313 -4.65668e-10 0.00195313 -2.00003 0.00195313 -2.00003 0.00195313 -8.59006e+09 0.00195313 -8.59006e+09 0.00195313 -3.6894e+19 0.00195313 -3.6894e+19 0.00195313 -1.58459e+29 0.00195313 -1.58459e+29 0.00195313 1.77965e-43 0.00195313 1.77965e-43 0.00195313 2.52439e-29 0.00195313 2.52439e-29 0.00195313 1.08422e-19 0.00195313 1.08422e-19 0.00195313 4.65668e-10 0.00195313 4.65668e-10 0.00195313 2.00003 0.00195313 2.00003 0.00195313 8.59006e+09 0.00195313 8.59006e+09 0.00195313 3.6894e+19 0.00195313 3.6894e+19 0.00195313 1.58459e+29 0.00195313 1.58459e+29 0.00195313 -1.77965e-43 0.00195313 -1.77965e-43 0.00195313 -2.52439e-29 0.00195313 -2.52439e-29 0.00195313 -1.08422e-19 0.00195313 -1.08422e-19 0.00195313 -4.65668e-10 0.00195313 -4.65668e-10 0.00195313 -2.00003 0.00195313 -2.00003 0.00195313 -8.59006e+09 0.00195313 -8.59006e+09 0.00195313 -3.6894e+19 0.00195313 -3.6894e+19 0.00195313 -1.58459e+29 0.00195313 -1.58459e+29 0.00195313 1.77965e-43 0.00195313 1.77965e-43 0.00195313 2.52439e-29 0.00195313 2.52439e-29 0.00195313 1.08422e-19 0.00195313 1.08422e-19 0.00195313 4.65668e-10 0.00195313 4.65668e-10 0.00195313 2.00003 0.00195313 2.00003 0.00195313 8.59006e+09 0.00195313 8.59006e+09 0.00195313 3.6894e+19 0.00195313 3.6894e+19 0.00195313 1.58459e+29 0.00195313 1.58459e+29 0.00195313 -1.77965e-43 0.00195313 -1.77965e-43 0.00195313 -2.52439e-29 0.00195313 -2.52439e-29 0.00195313 -1.08422e-19 0.00195313 -1.08422e-19 0.00195313 -4.65668e-10 0.00195313 -4.65668e-10 0.00195313 -2.00003 0.00195313 -2.00003 0.00195313 -8.59006e+09 0.00195313 -8.59006e+09 0.00195313 -3.6894e+19 0.00195313 -3.6894e+19 0.00195313 -1.58459e+29 0.00195313 -1.58459e+29 0.00195313 1.77965e-43 0.00195313 1.77965e-43 0.00195313 2.52439e-29 0.00195313 2.52439e-29 0.00195313 1.08422e-19 0.00195313 1.08422e-19 0.00195313 4.65668e-10 0.00195313 4.65668e-10 0.0 [... cut ...] 2.00003 0.00195313 8.59006e+09 0.00195313 8.59006e+09 0.00195313 3.6894e+19 0.00195313 3.6894e+19 0.00195313 1.77965e-43 0 0 0 1.0842e-19 0.00213623 1.77965e-43 0 2.35099e-38 0 0 7.39886e-43 0 7.39886e-43 0 0 0 0 -2.58494e-26 2.8026e-45 0 0 3.58732e-43 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 2 7.51707e+31 1.1911e-43 0 0 0 ...
```
and then the program exits with a non-zero code, with no other error output. Is it something wrong with the .yaml file, or something else?

jamestkpoon, 2022-02-14:
I should add that adding
```
Camera2.fx: 522.6021728515625
Camera2.fy: 522.6021728515625
Camera2.cx: 650.2871704101562
Camera2.cy: 359.57666015625
```
under Camera1 has no noticeable effect.

HouBiaoLiu, 2022-03-04:
src/Settings.cc
output << "" << ": [";
for(size_t i = 0; i < settings.originalCalib2_->size(); i++){
output << " " << settings.originalCalib2_->getParameter(i);
}
output << " ]" << endl;
add a condition if (settings.cameraType_ != Settings::Rectified)

jamestkpoon, 2022-03-09:
Thanks for the reply, @HouBiaoLiu . Seems to help, but now the program exits non cleanly after
```
Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
```