UZ-SLAMLab/ORB_SLAM3 #497: Running the test dataset nothing happens nor any errors

3026187875, 2022-02-20:
leioukupo@leioukupo-win:~/ORB_SLAM3$ ./Examples/Monocular/mono_euroc ./Vocabulary/ORBvoc.txt ./Examples/Monocular/EuRoC.yaml ../DataSheet/MH01 ./Examples/Monocular/EuRoC_TimeStamps/MH01.txt dataset-MH01_mono
num_seq = 1
file name: dataset-MH01_mono
Loading images for sequence 0...LOADED!

-------

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular
Loading settings from ./Examples/Monocular/EuRoC.yaml
Camera1.k3 optional parameter does not exist...
        -Loaded camera 1
        -Loaded image info
        -Loaded ORB settings
Viewer.imageViewScale optional parameter does not exist...
        -Loaded viewer settings
System.LoadAtlasFromFile optional parameter does not exist...
System.SaveAtlasToFile optional parameter does not exist...
        -Loaded Atlas settings
System.thFarPoints optional parameter does not exist...
        -Loaded misc parameters
----------------------------------
SLAM settings:
        -Camera 1 parameters (Pinhole): [ 458.65399169921875 457.29598999023438 367.21499633789062 248.375 ]
        -Camera 1 distortion parameters: [  -0.28340810537338257 0.073959067463874817 0.00019359000725671649 1.7618711353861727e-05 ]
        -Original image size: [ 752 , 480 ]
        -Current image size: [ 600 , 350 ]
        -Camera 1 parameters after resize: [  365.94735717773438 333.44500732421875 292.99069213867188 181.10678100585938 ]
        -Sequence FPS: 20
        -Features per image: 1000
        -ORB scale factor: 1.2000000476837158
        -ORB number of scales: 8
        -Initial FAST threshold: 20
        -Min FAST threshold: 7

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name:
There are 1 cameras in the atlas
Camera 0 is pinhole
First KF:0; Map init KF:0
New Map created with 271 points

Shutdown

Saving trajectory to f_dataset-MH01_mono.txt ...
There are 1 maps in the atlas
  Map 0 has 279 KFs

End of saving trajectory to f_dataset-MH01_mono.txt ...

Saving keyframe trajectory to kf_dataset-MH01_mono.txt ...

How to fix it ?

zlx1568, 2022-02-21:
I also encountered this problem, how can I solve it?

BystanderJ, 2022-02-22:
Same to me. Can somebody solve it?

zlx1568, 2022-02-22:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

3026187875, 2022-02-27:
I solved it. My opencv is 4.5.0, when i changed to 4.4.0, it worked

scottymax, 2022-04-14:
> I had the same problem. It might be hanging because of missing files or configuration. When I try the mono-inertial script, it works fine.
> 
> `$ ./Examples/Monocular-Inertial/mono_inertial_euroc Vocabulary/ORBvoc.txt Examples/Monocular-Inertial/EuRoC.yaml /Datasets/Euroc/ Examples/Monocular-Inertial/EuRoC_TimeStamps/MH01.txt`
> 
> The Datasets directory is obviously something you need to provide. It's weird that monocular just doesn't work though.

You can check the file **mono_euroc.cc** ,line 83.Change the last parameter to true.The viewer function should work after run build.sh.
`ORB_SLAM3::System SLAM(argv[1],argv[2],ORB_SLAM3::System::MONOCULAR, **false**);`

Redhwan-A, 2024-11-15:
Try this one:

`./Examples/Monocular-Inertial/mono_inertial_euroc ./Vocabulary/ORBvoc.txt ./Examples/Monocular-Inertial/EuRoC.yaml ~/Datasets/EuRoC/MH01 ./Examples/Monocular-Inertial/EuRoC_TimeStamps/MH01.txt dataset-MH01_monoimu`