UZ-SLAMLab/ORB_SLAM3 #499: When I tested with own data, reset occurs frequently. (Mono-Inertial)

YouMinJung, 2022-02-24:
Hi.
when I tested with EuRoC dataset, I get the good result.
So I want to test with own data.

I collected camera images and imu data with timestamp.
My dataset is like below.
[image]

Furthermore, I calibrated camera-imu with kalibr.
This is the own YAML file.
[image]

However, I tested with orb-slam (mono-inertial), remapping occured very frequently.
Reset occurred approximately every 4 seconds...

What is the problem?
How to solve this problem..?? help..

makolele12, 2022-02-25:
Are you sure your NoiseGyro value is okay? It is way to big in comparison to the other 3 parameters.

I am having the exact same behavior in Monocular-Inertial. I am using the Intel RealSense D435i camera and even when using the calibration parameters they propose, the IMU does not get initialized and the map is frequently being reinitialized.

This does not happen to me with the same camera and videos but in Monocular-only though.

YouMinJung, 2022-03-02:
I calibrated imu sensor using imu_utils.
When I record the imu data for the calibration, I shook the sensor approximately 45 minutes.
Is this correct action?

Maithilishetty, 2022-03-13:
Even I am facing frequent reset problem. Did anyone solve this?

ynma-hanvo, 2022-03-31:
same for me.  I used stereo-Inertial  with my stereo cameras, parameter from kalibr result. 
also tracking reset every few seconds.   for stereo is ok[  segement fault after running some time although]

rancheng, 2022-04-20:
have a look here guys:

 [issue #280](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/issues/280#issuecomment-478247453)

you probably calibrated the `noise density` with `kalibr`, please make sure you convert it back to `noise`!

Always pay attention to the details.

sandstorm666, 2022-05-12:
@rancheng 
I did what you said and still had frequent resets. I calibrated the parameters of the imu as follows.
IMU0:
  Model: calibrated
  Update rate: 200.0
  Accelerometer:
    Noise density: 0.0181066131887 
    Noise density (discrete): 0.2560661794 
    Random walk: 0.000479934988114
  Gyroscope:
    Noise density: 0.00191159599129
    Noise density (discrete): 0.0270340497667 
    Random walk: 1.34283856163e-05
This happens even though I multiply or divide by the square root.

rancheng, 2022-09-15:
@wwendy233 are you calibrating the imu with this tool: [imu_utils](https://github.com/gaowenliang/imu_utils) ?

mahammadirfan, 2022-09-20:
Hi @rancheng ,
I am using https://github.com/ori-drs/allan_variance_ros
Do i need to convert the unit as well ?

ricardo11235, 2022-10-24:
Hi @YouMinJung ，
Can you tell me what type of imu you're using？Does the fps of imu affect the algorithm？my imu can only output 75fps

LarryDong, 2023-03-26:
Hi guys,
I searched for the unit and gives the conclusion:
1. Try to use [https://github.com/ori-drs/allan_variance_ros](https://github.com/ori-drs/allan_variance_ros) but not [imu_utils](https://github.com/gaowenliang/imu_utils). Check [this discussion](https://discourse.ros.org/t/open-source-allan-variance-tool-for-rosbags/23136/6) for detials.
2. allan_variance_ros's output unit is just the same as orb-slam3, no convertion (like vins-mono, convert to discrete time) is needed. The unit is just continous form, which I think @rancheng 's comment is wrong. 
3. Reset frequently comes from other issues, not just the unit.