UZ-SLAMLab/ORB_SLAM3 #500: EdgeInertialGS Jacobians calculate issue

Lagrange1027, 2022-02-24:
Hi,all:
when I read ImuInitialize ，I found a issue in EdgeInertialGS Jacobians  calculate , please hellp me make a check ,thanks!
[image]

I think maybe the result should look like this 
[image]

Lagrange1027, 2022-02-24:
sorry，I know the reason， It‘s my fault

[image]

khudkhud, 2022-03-26:
It also confused me.

khudkhud, 2022-03-26:
@Lagrange1027 请教您一下：
后来的推导，增量为什么要乘以旋转矩阵呢？

zhh2005757, 2022-10-04:
> sorry，I know the reason， It‘s my fault
> 
> [image]

I think the jacobian w.r.t the translation of Pose 2 is not _exactly_ the $\bar{p}_i$ in the paper. It should be $p_i$ where $p_i=R^T_i * \bar{p}_j$ here. And that's what it is in the code.

Turing47, 2024-05-08:
I think the question is good, and code could be wrong.