UZ-SLAMLab/ORB_SLAM3 #501: black screen and "waiting for images" when trying mono with USB camera

esteimle, 2022-02-28:
Hi. I've been trying for a few days to get my USB camera to work with ORBSLAM3. I started with the mono_realsense_t265.cc as my base, made a copy of that file, and then added opencv cv::VideoCapture to connect to my USB camera. When I debug, I can see the frames coming in as well as the timestamps updating with each frame. I displayed some pictures as well to verify they are real. I also calibrated my camera and modified the t265 yaml file to use those parameters. Has anyone else hit something similar? My next steps are going to be digging into the code step by step but that's going to take a while ;)  Thank you for any advice.

**Command line output**
`~/workspace/ORB_SLAM3$ ./Examples/Monocular/mono_razor Vocabulary/ORBvoc.txt ./Examples/Monocular/Razor.yaml

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular
Loading settings from ./Examples/Monocular/Razor.yaml
Camera1.k1 optional parameter does not exist...
	-Loaded camera 1
Camera.newHeight optional parameter does not exist...
Camera.newWidth optional parameter does not exist...
	-Loaded image info
	-Loaded ORB settings
Viewer.imageViewScale optional parameter does not exist...
	-Loaded viewer settings
System.LoadAtlasFromFile optional parameter does not exist...
System.SaveAtlasToFile optional parameter does not exist...
	-Loaded Atlas settings
System.thFarPoints optional parameter does not exist...
	-Loaded misc parameters
----------------------------------
SLAM settings: 
	-Camera 1 parameters (Pinhole): [ 243.38832092285156 241.70191955566406 365.0584716796875 238.09236145019531 ]
	-Original image size: [ 640 , 480 ]
	-Current image size: [ 640 , 480 ]
	-Sequence FPS: 60
	-Features per image: 1000
	-ORB scale factor: 1.2000000476837158
	-ORB number of scales: 8
	-Initial FAST threshold: 20
	-Min FAST threshold: 7

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
There are 1 cameras in the atlas
Camera 0 is pinhole
Starting the Viewer
[ WARN:12] global /home/pradeep/workspace/opencv_build/opencv/modules/videoio/src/cap_gstreamer.cpp (956) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
`

**My Config file**
`%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------
File.version: "1.0"

Camera.type: "PinHole"

# Left Camera calibration and distortion parameters (OpenCV)
Camera1.fx: 243.38832282
Camera1.fy: 241.7019257
Camera1.cx: 365.05847057
Camera1.cy: 238.09236466

# Kannala-Brandt distortion parameters
#Camera1.k1: -0.00530046410858631
#Camera1.k2: 0.0423333682119846
[... 239 lines cut ...]

            // Pass the image to the SLAM system
            SLAM.TrackMonocular(imCV, timestamp_ms);

#ifdef REGISTER_TIMES
  #ifdef COMPILEDWITHC11
            std::chrono::steady_clock::time_point t2 = std::chrono::steady_clock::now();
  #else
            std::chrono::monotonic_clock::time_point t2 = std::chrono::monotonic_clock::now();
  #endif
            t_track = t_resize + std::chrono::duration_cast<std::chrono::duration<double,std::milli> >(t2 - t1).count();
            SLAM.InsertTrackTime(t_track);
#endif

        //}
    }

    //pipe.stop();

    // Stop all threads
    SLAM.Shutdown();

    return 0;
}
`

Maithilishetty, 2022-03-02:
@esteimle I am having a similar issue while running the examples on ROS. Have you managed to fix this yet?

esteimle, 2022-03-02:
yeah, just an a hour ago.  I lowered the resolution to 640x480 in the yaml file, then also lowered it in the opencv capture (not resizing it later but setting the camera to input 640x480).  It seemed to get lost pretty easy but it made a 1/4 map of my office :)  I think the high res input from my camera was bogging down the opencv camera capture.  Either that or I had my resolutions misaligned between yaml and real life.

closed 2022-03-02 by esteimle

Maithilishetty, 2022-03-02:
Oh great, thanks for the response! I do not believe that's a possible issue for me since I am just running the EuRoC rosbag file from the examples section. I am not sure how to fix it.