UZ-SLAMLab/ORB_SLAM3 #503: build.sh errors

AlanMohammad20, 2022-03-01:
OS : ubuntu 20.04
ROS noeitc
after check for all prerequisites
try to build ORB_SLAM3
and face this error 

Configuring and building Thirdparty/DBoW2 ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/alan/gitHub/ORB_SLAM3/Thirdparty/DBoW2/build
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: cannot create directory ‘build’: File exists
-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/alan/gitHub/ORB_SLAM3/Thirdparty/g2o/build
[100%] Built target g2o
Configuring and building Thirdparty/Sophus ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/alan/gitHub/ORB_SLAM3/Thirdparty/Sophus/build
[ 16%] Built target test_so2
[ 16%] Built target test_common
[ 25%] Built target test_rxso2
[ 33%] Built target test_geometry
[ 41%] Built target test_sim2
[ 50%] Built target test_velocities
[ 58%] Built target test_so3
[ 66%] Built target test_sim3
[ 75%] Built target test_se2
[ 83%] Built target test_se3
[ 91%] Built target HelloSO3
[100%] Built target test_rxso3
Uncompress vocabulary ...
Configuring and building ORB_SLAM3 ...
mkdir: cannot create directory ‘build’: File exists
Build type: Release
-- Using flag -std=c++14.
OPENCV VERSION:
4.4.0
-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/alan/gitHub/ORB_SLAM3/build
[ 20%] Built target g2o
make[2]: *** No rule to make target '/home/alan/gitHub/Pangolin/build/libpango_glgeometry.so', needed by '../lib/libORB_SLAM3.so'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:971: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

any one can help please

astronaut71, 2023-06-19:
have the same issue. Have you solved?

yogurtbuddy, 2024-01-21:
> have the same issue. Have you solved?

the same