UZ-SLAMLab/ORB_SLAM3 #510: Executing /Examples/mono_euroc.cc does not open up Pangolin visualizations?

Mechazo11, 2022-03-10:
Dear Authors,

Good afternoon. Thank you very much for supporting open-source research on localization and mapping of mobile robots by making your latest version of ORB-SLAM3 open-source.

I plan on using this system as a benchmarking tool against a new method we are working on.

To understand how the system ingests data to produce VO estimates we are using the MH01 dataset from EuRoc. However when we invoke the command [Note, the Dataset directory was placed in the /home directory for convenience]

```
./Examples/Monocular/mono_euroc ./Vocabulary/ORBvoc.txt ./Examples/Monocular/EuRoC.yaml ~/Datasets/EuRoc/MH01/ ./Examples/Monocular/EuRoC_TimeStamps/MH01.txt mh1_out
```

I can see that the system is cycling through the frames in the image sequence (I activated a line that prints out the timestep) but no visualization window (one showing 3D movement and the other showing ORB keypoints) is coming up. 

Our system configuration is as follows
* Ubuntu 20.04
* ROS Neotic
* OpenCV 4.2.034 [I changed CMakelist.txt in ORB-SLAM3 to accept openCV 4.2]
* Pangolin [V0.6 with OpenNI and OpenNI2 library check disabled]
* EuRoC Dataset Machine Room 1 (MH01)

My questions are the following

1. Is there an argument/command that we pass in to activate Pangolin Visualization?
2. Does V1.0 of ORB-SLAM3 require OpenNI and OpenNI2 drivers?

Some notes
1. We were having a hard time installing Pangolin's latest version due to a library linking issue for libOpenNI2.so.0 file. Hence reverting to the older version and disabling library checks for 

Any help and suggestions in getting the visualization load up when the system works would be highly appreciated.

With best regards,
Azmyin Md. Kamal,
Ph.D. student, ICORE robotics lab, LSU
Baton Rouge, LA-70803

kegsay, 2022-03-20:
https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/Examples/Monocular/mono_euroc.cc#L83

The last arg needs to be `true` as that sets `bUseViewer` - https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/src/System.cc#L42

ogamache, 2022-06-09:
I had the same problem. Solved it with @kegsay's method, thanks!