UZ-SLAMLab/ORB_SLAM3 #511: Running ORB-SLAM3 with D435i . waiting for images

sunxirui310, 2022-03-11:
Thank you for your work. 
use intel realsense D435i. I execute the following command in terminal ,like the examples "4. Running ORB-SLAM3 with your camera"
`./Examples/RGB-D/rgbd_realsense_D435i Vocabulary/ORBvoc.txt ./Examples/RGB-D/RealSense_D435i.yaml 666` 
or
`./Examples/Monocular/mono_realsense_D435i Vocabulary/ORBvoc.txt ./Examples/Monocular/RealSense_D435i.yaml 777`
or other tests command. I did not modify the provided “ d435i yaml” file in each example folder.

it always shows "waiting for images"
[image]

I can see via "realsense viewer" that the camera is working
[image]

Ubuntu 20.04 ,I build and make install with compiler c++11:
Pangolin 0.5
OpenCV 4.4
Eigen 3.4
realsenseSDK 2.5.0

Can someone give some point？

sunxirui310, 2022-03-11:
EuRoC dataset is OK.
`./Examples/Monocular-Inertial/mono_inertial_euroc ./Vocabulary/ORBvoc.txt ./Examples/Monocular-Inertial/EuRoC.yaml /home/yfb/work/data/EuRoC/MH_01_easy ./Examples/Monocular-Inertial/EuRoC_TimeStamps/MH01.txt dataset-MH01_monoi`
[image]

JACKLiuDay, 2022-03-11:
I have the same problem with you

sunxirui310, 2022-03-11:
> I have the same problem with you

TAI CAN LE！哭晕在厕所

anchao1997, 2022-03-13:
I have the same problem with you,Do you solve it?

nanguaxs, 2022-03-16:
have you solved it now? I have the same problem, I have dealt with it for a long time,555 哭死了

ryleymcc, 2022-03-18:
When you get the "WAITING FOR IMAGES" did you try to move the camera around?

JACKLiuDay, 2022-03-19:
> When you get the "WAITING FOR IMAGES" did you try to move the camera around?

yes , i have a try and it still shows "WATING FOR IMAGES"
I built the orb at nvidia NX

ryleymcc, 2022-03-19:
Have you run stereo only?

sunxirui310, 2022-03-21:
> Have you run stereo only?

i ran mono / rgb-d /rgbd+imu /mono+imu
and move camera around 

always shows "WATING FOR IMAGES" 

i'm tring to add some code like "cv::imshow","std::cout" to find where the problem is .

sunxirui310, 2022-03-22:
我D435i调通了。
Ubuntu20.04，笔记本（surface重装了ubuntu）和台式机均好用。
建议D435i有问题的同学再去检查一下librealsense的安装。
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

realsense-viewer能打开相机并不意味着librealsense安装成功。
详细对一下里边的每一步，是否安装了需要的环境apt install XXX，以及执行了相应的脚本 ./script/中的 XXX.sh
例如 ./scripts/setup_udev_rules.sh等。

rebuild Realsense后，记得rebuild ORBSLAM3。

JACKLiuDay, 2022-03-22:
> 我D435i调通了。 Ubuntu20.04，笔记本（surface重装了ubuntu）和台式机均好用。 建议D435i有问题的同学再去检查一下librealsense的安装。 https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md [...]

Bro, have you ever try this on nvidia NX? I build it on the NX platform and it shows "WAITING FOR IMAGES"

sunxirui310, 2022-03-22:
> Bro, have you ever try this on nvidia NX? I build it on the NX platform and it shows "WAITING FOR IMAGES"

i don't have nvidia jetson。sorry

JACKLiuDay, 2022-03-22:
> i don't have nvidia jetson。sorry

Thank you anyway , bro

anchao1997, 2022-03-22:
> 我D435i调通了。 Ubuntu20.04，笔记本（surface重装了ubuntu）和台式机均好用。 建议D435i有问题的同学再去检查一下librealsense的安装。 https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md [...]

请问你是手动从源码安装librealsense的吗？

sunxirui310, 2022-03-22:
> 

是的，担心分发包版本问题。于是乎自己build

anchao1997, 2022-03-22:
> > 
> 
> 是的，担心分发包版本问题。于是乎自己build

好的谢谢

ynma-hanvo, 2022-03-31:
> > 
> 
> 是的，担心分发包版本问题。于是乎自己build

how is the result with realsense cameras.   does tracking lost and reset often. 
and   is the vio path accurate?

sunxirui310, 2022-03-31:
> > > 
> > 
> > 
> > 是的，担心分发包版本问题。于是乎自己build
> 
> how is the result with realsense cameras. does tracking lost and reset often. and is the vio path accurate?

D435i测试以下几个场景（RGB-D）：
1走廊过暗时，无ORB特征点，系统卡住。
2太阳光过亮，也会检测拍摄物体的边界、角点特征，导致路径估计差的比较多。
3相机拍摄白色墙面时，无法检测到特征点也会丢失当前状态。
4其他场景基本可正常检测特征点。

实验收获
1.多数情况下，我认为效果并不如意，从特征点检测、匹配到路径跟踪，都不稳。
2.在转弯的时候，经常出现跟踪丢失的情况。得慢点转
3.围绕一个实验场景转多个圈，路径估计比只转一圈准一些。
4.摄像头高度过低会严重影响路径估计准确度。
5.直线相对比较准。
6.rgbd+imu我的程序变得很卡顿，暂时只整vo，不包含imu部分。

有些疑问还没想明白，例如
1.D435i 有自标定（self-calibration）程序，相机内参准确程度对最后的路线估计有多大影响？
2.做实验如何知道真值（Ground Truth）？包括路径和每个时刻的相机姿态。

另外我把源码ORBextracor.cc，ORBextracor.h单独拿出来做了对比实验，感觉效果并没有比opencv的算法更好，时间和效果都是。
v-slam这块刚接触，还在学习积累和优化探索阶段，可能实验过程也有问题。还请各位大佬多多指点。

sunxirui310, 2022-03-31:
D435i tests the following scenarios (RGB-D):
1 When the corridor is too dark, there is no ORB feature point, and the system is stuck.
2 When the sun is too bright, the boundary and corner features of the photographed object will also be detected, resulting in poor path estimation.
3 When the camera shoots a white wall, the feature points cannot be detected and the current state will be lost.
4 Other scenes can basically detect feature points normally.

Experimental harvest
1. In most cases, I think the effect is not satisfactory, from feature point detection, matching to path tracking, it is unstable.
2. When turning, the situation of tracking loss often occurs. turn slowly
3. Make multiple circles around an experimental scene, and the path estimation is more accurate than just making one circle.
4. If the camera height is too low, it will seriously affect the path estimation accuracy.
5. The straight line is relatively accurate.
6. rgbd+imu My program has become very stuck, temporarily only the whole vo, not including the imu part.

There are some questions that I haven't figured out yet, such as
1. The D435i has a self-calibration program. How much influence does the accuracy of the camera's internal parameters have on the final route estimation?
2. How to know the ground truth by doing experiments? Including the path and the camera pose at each moment.

In addition, I took out the source code ORBextracor.cc and ORBextracor.h to do a comparative experiment. I feel that the effect is not better than the opencv algorithm, both in time and effect.
The v-slam is just new, and it is still in the learning accumulation and optimization exploration stage. There may be problems in the experimental process. Also please give more pointers.

jinshuitiaofan, 2022-05-07:
> EuRoC dataset is OK. `./Examples/Monocular-Inertial/mono_inertial_euroc ./Vocabulary/ORBvoc.txt ./Examples/Monocular-Inertial/EuRoC.yaml /home/yfb/work/data/EuRoC/MH_01_easy ./Examples/Monocular-Inertial/EuRoC_TimeStamps/MH01.txt dataset-MH01_monoi` [image]

你好，那个我./build.sh之后，就没有mono_realsense_D435i这个二进制执行文件，只有同名的.cc文件，我想请问下你是怎么编译执行的，我下载的ORB_SLAM3是master。

goodstudent9, 2024-05-16:
> 你好，那个我./build.sh之后，就没有mono_realsense_D435i这个二进制执行文件，只有同名的.cc文件，我想请问下你是怎么编译执行的，我下载的ORB_SLAM3是master。

我有同样的问题

nanguaxs, 2024-05-19:
这是几年前的工作了，都忘了，抱歉呀

goodstudent9, 2024-05-19:
已经解决了，是因为没有装realsense的lib

 这是几年前的工作了，都忘了，抱歉呀

nanguaxs, 2024-05-27:
好的好的