UZ-SLAMLab/ORB_SLAM3 #514: ros node

anchao1997, 2022-03-13:


anchao1997, 2022-03-13:
while I am running ros node with intel realsense d435i,
rosrun ORB_SLAM3 Stereo_Inertial Vocabulary/ORBvoc.txt Examples/Stereo-Inertial/RealSense_D435i.yaml false /camera/left/image_raw:=/camera/infra1/image_rect_raw /camera/right/image_raw:=/camera/infra2/image_rect_raw /imu:=/camera/imu
and the .yaml has been changed like this:
%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------
Camera.type: "PinHole"

# Left Camera calibration and distortion parameters (OpenCV)
Camera.fx: 382.613
Camera.fy: 382.613
Camera.cx: 320.183
Camera.cy: 236.455

# distortion parameters
Camera.k1: 0.0
Camera.k2: 0.0
Camera.p1: 0.0
Camera.p2: 0.0

# Right Camera calibration and distortion parameters (OpenCV)
Camera2.fx: 382.613
Camera2.fy: 382.613
Camera2.cx: 320.183
Camera2.cy: 236.455

# distortion parameters
Camera2.k1: 0.0
Camera2.k2: 0.0
Camera2.p1: 0.0
Camera2.p2: 0.0

Tlr: !!opencv-matrix
  rows: 3
  cols: 4
  dt: f
  data: [1.0, 0.0, 0.0, 0.0499585,
         0.0, 1.0, 0.0, 0.0,
         0.0, 0.0, 1.0, 0.0,
         0.0, 0.0, 0.0, 1.0]

# Camera resolution
Camera.width: 640
Camera.height: 480

# Camera frames per second 
Camera.fps: 30.0

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

# Image scale, it changes the image size to be processed (<1.0: reduce, >1.0: increase)
Camera.imageScale: 0.5 # 0.5 #0.7071 # 1/sqrt(2)

# Close/Far threshold. Baseline times.
ThDepth: 40.0
# stereo baseline times fx
Camera.bf: 19.114771561

# Transformation from body-frame (imu) to left camera
Tbc: !!opencv-matrix
   rows: 4
   cols: 4
   dt: f
   data: [1,0,0,-0.005,
         0,1,0,-0.005,
         0,0,1,0.0117,
         0.0, 0.0, 0.0, 1.0]

# Do not insert KFs when recently lost
InsertKFsWhenLost: 0

# IMU noise (Use those from VINS-mono)
IMU.NoiseGyro: 1e-3 # 2.44e-4 #1e-3 # rad/s^0.5
IMU.NoiseAcc: 1e-2 # 1.47e-3 #1e-2 # m/s^1.5
IMU.GyroWalk: 1e-6 # rad/s^1.5
IMU.AccWalk: 1e-4 # m/s^2.5
IMU.Frequency: 200

#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------
# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 1250

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.2

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 8

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#--------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 0.9
Viewer.PointSize: 2
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -3.5
Viewer.ViewpointF: 500

and the throws exception 
There are 1 cameras in the atlas
Camera 0 is pinhole
Starting the Viewer
**Empty IMU measurements vector!!!
terminate called after throwing an instance of 'std::system_error'
  what():  Operation not permitted
Aborted (core dumped)**

and with this phenomenon occurs
[image]
the View flashed
and my Pangolin version is 0.6 release
CAN someone help me?

anchao1997, 2022-03-13:
`~/ORB_SLAM3$ ./Examples/Stereo-Inertial/stereo_inertial_realsense_D435i Vocabulary/ORBvoc.txt ./Examples/Stereo-Inertial/RealSense_D435i.yaml`
the same thing occurs,but this time there is no viewer 
`There are 1 cameras in the atlas
Camera 0 is pinhole
146 dropped frs
terminate called after throwing an instance of 'std::system_error'
  what():  Operation not permitted
Aborted (core dumped)`

lawpdas, 2022-03-24:
same problem

anchao1997, 2022-03-26:
> same problem

I finally found I use the false config(RealSense_D435i.yaml),if you are running ros node you should use the RealSense_D435i.yaml of Examples_old

closed 2022-04-03 by anchao1997

silencht, 2023-12-14:
May I ask which parameters of the .yaml configuration file cause this error?