UZ-SLAMLab/ORB_SLAM3 #515: Question about localization mode

TeerapongPoom, 2022-03-14:
Hello. I have tried using load a saved map and use it with localization mode in RGBD.
Rosbag I use to test is same one I used to create map. 
Config file is the same as when I create map, just remove System.SaveAtlasToFile and add System.LoadAtlasFromFile.
At first, it does not localize at all and it keep showing

> TRACK_REF_KF: Less than 15 matches!!

in the terminal. Here is the screencap
[https://drive.google.com/file/d/1JyahqoTWlvA14VDXyPWwT3xhMFKTOe38/view?usp=sharing](url)

So I dig in a bit and add `mpReferenceKF->ComputeBoW();`
to function `void Tracking::StereoInitialization()` in Tracking.cc

This time orb can localize itself to some degree as shown in the video
[https://drive.google.com/file/d/1PVNlC3dkZiw3BacLtpmyCva_2WRh8R1q/view?usp=sharing](url)
But camera pose green frame still diff a lot from blue frame even though it is the same image sequence.

So I change code in System.cc line 171 and force atlas to set current map to loaded map 
from 
`mpAtlas->CreateNewMap();` 
to 
`vector<Map*> map_vector = mpAtlas->GetAllMaps();`
`mpAtlas->ChangeMap(map_vector.at(0));`

It seems to localize itself pretty well 
[https://drive.google.com/file/d/18AJXIBckYQSV-LkoaeMRzeF2dquyR7IU/view?usp=sharing](url)
And still work when i change to another sequence of the same room
[https://drive.google.com/file/d/1fGL-b8dGZ9yC64JTEyZN1rLAIahdqoH6/view?usp=sharing](url)

However when I start half way of the sequence to make image start at middle of the room, orb did not localize itself to position where it should be. And shown some strange behavior soon after. 
[https://drive.google.com/file/d/1Y761xWXlHhAPUBrDyw79hJZ_04hCzJR1/view?usp=sharing](url)

My intention is to use depth camera as range sensor to localize in crowded environment. So I try to use orbs in slam mode and create map in empty environment first and then use localization mode in crowded environment later.
Or my understanding about localization mode is wrong and it's cannot be used this way?

lihzh233, 2022-04-06:
I check the function tracking::track() in Tracking.cc, and find that if you work in localization mode, your mState will never turn into LOST. But in line 2039, it call Relocalization() under the condition of mState==LOST. Simply change the condition from `if (mState == LOST)` to `if (mState == LOST || mState == RECENTLY_LOST)` may help.

JACKLiuDay, 2022-04-12:
Hi, bro, I have the same intention with. I want to build a map with ORB first, then use it to localize. Now I have a big problem. When the atlas map loaded from file , the orb still try to make new map , it does not use the map I loaded. So strange. Do you have any ideas.

JACKLiuDay, 2022-04-18:
> I check the function tracking::track() in Tracking.cc, and find that if you work in localization mode, your mState will never turn into LOST. But in line 2039, it call Relocalization() under the [...]

Hi guy, I see the function in tracking.cc 2039 line. 
[image]
I see the TODO Not available in inertial mode.
Dose it mean localization not work with imu-inertial or imu-stereo mode?

wildpig888, 2022-05-19:
I tried to save and load the map. I found that localization mode made the system more sluggish than SLAM mode. After a few minutes the computer froze. Have you ever noticed this situation?

JACKLiuDay, 2022-05-20:
> 

I tried with the save and load function too. It has many problem. There is no good solution so far.

tungtungyan, 2022-06-14:
> > 
> 
> I tried with the save and load function too. It has many problem. There is no good solution so far.

How can I use the load function? I add System.LoadAtlasFromFile into the YAML file but it didn’t work.

Shang-cl, 2023-06-09:
@TeerapongPoom 
Hi, I am now trying to use the binocular + IMU method to test the positioning mode. First, I used the binocular + IMU to successfully save and load the map, and according to your idea, I realized the binocular pure vision. Pure positioning mode, but when I want to use binocular + IMU to achieve the same function, I find it seems impossible. Do you have any better solutions or suggestions? Thanks!