UZ-SLAMLab/ORB_SLAM3 #517: ORB_SLAM3 - Pangolin Visualization not showing

ayushGHub, 2022-03-16:
Pangolin Visualization is not showing in UBUNTU 20.04. However, everything else is working fine and the calculation is also being done and the trajectory file is being calculated correctly.

```

Saving trajectory to f_dataset-MH01_monoTRIAL_with_Pangolin5.txt ...
There are 1 maps in the atlas
  Map 0 has 327 KFs

End of saving trajectory to f_dataset-MH01_monoTRIAL_with_Pangolin5.txt ...

Saving keyframe trajectory to kf_dataset-MH01_monoTRIAL_with_Pangolin5.txt ...
```

I also installed all the dependencies from [Pangolin Official Website](https://github.com/stevenlovegrove/Pangolin/tree/ad8b5f83222291c51b4800d5a5873b0e90a0cf81)

There is no error at all.

lihzh233, 2022-03-16:
`ORB_SLAM3::System SLAM(argv[1],argv[2],ORB_SLAM3::System::MONOCULAR, true);`
check this code, the last param should be 'true'.

ayushGHub, 2022-03-16:
thanks, it's working fine now.

zhangletian2, 2022-03-17:
I tried it on wsl2 , but it didn't work for me,any other solutions?

Water1775, 2022-03-17:
你好，请问您使用过怎样的修改使他继续工作的？

foemre, 2022-08-01:
How exactly is this problem solved? Which tag have you tested with? Where is `ORB_SLAM3::System SLAM(argv[1],argv[2],ORB_SLAM3::System::MONOCULAR, true);` or any other substring located? Because as far as I can see there is no such line in the latest version of the code. This issue has not been resolved yet, can someone help?

foemre, 2022-08-01:
Ok so this issue is solved in this pull request: https://github.com/UZ-SLAMLab/ORB_SLAM3/pull/606