UZ-SLAMLab/ORB_SLAM3 #521: How to load/save map functionalities and missing scripts and time analysis

StarFyx, 2022-03-19:
Thank you for your open source code and excellent work. However I have some question about ORB_SLAM3V1.0 .

1. I wonder know how to load or save map in ORB_SLAM3 v1.0 which is mentioned in the release V1.0.
2. There is missing scripts such as `./tum_vi_examples` mentioned in the README.
3. There isn't line `#define REGISTER_TIMES` in `include\Config.h` , but some empty config class about IMU, viewer and camera.

Looking forward for reply , thank you very much.

JACKLiuDay, 2022-04-21:
Hi guy, if you want to know how to save/load map.Maybe you can look this issue. Some prompts are listed in it.
https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/443

StarFyx, 2022-04-21:
> Hi guy, if you want to know how to save/load map.Maybe you can look this issue. Some prompts are listed in it. #443