UZ-SLAMLab/ORB_SLAM3 #525: Why is the score about 5.991 but not 3.841 in the function CheckFundamental?

NiuKeke, 2022-03-22:
In the file named TwoViewReconstruction.cc, the 446th line code shows that it uses 5.991 threshold but not 3.841 when esmitate the F score .According to the chi-square distribution, we hope that it could be 95% accuracy with the 3.841 but not 5.991. So where are the problem? Maybe, i have understood it wrongly.Thank you very much!

makolele12, 2022-03-23:
I think the following line in the paper answers your question:

> &Gamma; is defined equal to T_H so that
> both models score equally for the same d in their inlier
> region, again to make the process homogeneous.

NiuKeke, 2022-03-30:
> I think the following line in the paper answers your question:
> 
> > Γ is defined equal to T_H so that
> > both models score equally for the same d in their inlier
> > region, again to make the process homogeneous.

Thank you very much ,i got it.