UZ-SLAMLab/ORB_SLAM3 #526: Some frames missing in the saved trajectory

NickPerezCarletonUniversity, 2022-03-23:
I configured Orb Slam to automatically save the full trajectory after doing a test run using data from a rosbag, and I notice that some frames are missing. For example, the rosbag I am using may have 4000 video frames, but the saved trajectory has 3900 frames only. I am using a stereo camera (no IMU data)

Is this normal? there is no indication while running that 'tracking failed' or any other error like that. I imagine there is just a few frames once in a while that don't track properly, but tracking does not completely fail. Thus some frames are missing without total failure of the SLAM algorithm.

lawpdas, 2022-04-13:
The same problem. I run monocular_tum on my own data and use SaveTrajectoryEuRoC to save camera trajectory. But there are only 328 frames have been saved for a 1000 frames video.
```
    // Save camera trajectory
    SLAM.SaveKeyFrameTrajectoryTUM(string(argv[3])+"ORB_SLAM3_KeyFrameTrajectory.txt");
    SLAM.SaveTrajectoryEuRoC(string(argv[3])+"ORB_SLAM3_CameraTrajectory.txt");  
```

NickPerezCarletonUniversity, 2022-04-13:
I tried running the same data again, but on a stronger computer. It saved more frames when running on a stronger machine.