UZ-SLAMLab/ORB_SLAM3 #531: Problems with Gazebo Cameras

Migran99, 2022-03-28:
Hello everyone!

I am having some issues trying to connect a simulated camera in Gazebo with the ORB SLAM 3. It just seems like the images are not being received by the slam program (I have checked topic names, messages types and so on and that's not the problem). It stays in "waiting for images".
In fact when I publish to the same topic but with my webcam with usb_cam and rosbags of datasets and it works perfectly.
So it seems that the problem might be with Gazebo but I don't know what might be wrong. The camera paremeters are these:
```	
<gazebo reference="camera_link">
		<sensor type="camera" name="camera">
			<update_rate>30.0</update_rate>
			<camera name="head">
				<horizontal_fov>1.3962634</horizontal_fov>
				<image>
					<width>640</width>
					<height>480</height>
					<format>R8G8B8</format>
				</image>
				<clip>
					<near>0.02</near>
					<far>300</far>
				</clip>
				<noise>
					<type>gaussian</type>
					<!-- Noise is sampled independently per pixel on each frame.
		       That pixel's noise value is added to each of its color
		       channels, which at that point lie in the range [0,1]. -->
					<mean>0.0</mean>
					<stddev>0.007</stddev>
				</noise>
			</camera>
			<plugin name="camera_controller" filename="libgazebo_ros_camera.so">
				<alwaysOn>true</alwaysOn>
				<updateRate>0.0</updateRate>
				<cameraName>camera</cameraName>
				<imageTopicName>image_raw</imageTopicName>
				<cameraInfoTopicName>camera_info</cameraInfoTopicName>
				<frameName>camera_link</frameName>
				<hackBaseline>0.7</hackBaseline>
				<distortionK1>0.0</distortionK1>
				<distortionK2>0.0</distortionK2>
				<distortionK3>0.0</distortionK3>
				<distortionT1>0.0</distortionT1>
				<distortionT2>0.0</distortionT2>
				<CxPrime>0</CxPrime>
				<Cx>0.0</Cx>
				<Cy>0.0</Cy>
				<focalLength>0.0</focalLength>
			</plugin>
		</sensor>
	</gazebo>
```
And the problem is that it freezes like that. Sometimes (randomly) it works for a frame and displays it:
[image]

Anyone with experience with this and would kindly help a noobie here?

Thank you very much!!

zazgf, 2022-03-30:
Maybe the timestamp reason

closed 2022-04-09 by Migran99

dionkrith, 2023-07-08:
I face the same problem. Any solution?

AirLover, 2024-03-26:
hi have you solved the problem?i have got the same question