UZ-SLAMLab/ORB_SLAM3 #533: TUM_1024  Monocular-Inertial Loop closing not work

zazgf, 2022-03-30:
@jdtardos Hi,Mr Juan D. Tardos.I just runned the TUM_1024  Monocular-Inertial example with the latest version. I found the loop closing not worked as expected, and I don't know why. Can you help me？
[image]

jdtardos, 2022-03-30:
You need at least 3 keyframes that see areas in common. Please, read the
ORB-SLAM3 paper.

Juan D. Tardos

zazgf, 2022-03-30:
@jdtardos Mr Juan D. Tardos, thanks for your advice. I will read the ORB_SLAM3 paper to understand this problem.

zazgf, 2022-04-04:
@jdtardos  Prof. Tardos, I have read the ORB_SLAM3 paper, and know I know your meaning , Thanks again!

closed 2022-04-04 by zazgf