UZ-SLAMLab/ORB_SLAM3 #534: dual fish eye or omni directional cameras

sadsasuke, 2022-03-30:
Hi, 
Is it possible to use omni directional cameras/ dual fish eye cameras with this ORB SLAM 3. 
Also there are two more followup questions.
1-  Which tools to use for calibration of such cameras
2- Can we use Juho kannala model of orb slam 3 with these cameras.

Looking forward for responses. Thanks

JasonTang84, 2022-04-15:
> Hi, Is it possible to use omni directional cameras/ dual fish eye cameras with this ORB SLAM 3. Also there are two more followup questions. 1- Which tools to use for calibration of such cameras 2- Can we use Juho kannala model of orb slam 3 with these cameras.
> 
> Looking forward for responses. Thanks

I am also looking for it. did u find any answers?

sadsasuke, 2022-04-21:
for now I am just using one side of the dual fish eye camera.

1-for calibration i used script from here
https://medium.com/@kennethjiang/calibrate-fisheye-lens-using-opencv-333b05afa0b0
2- Plug those parameters of K and D from above into the .yaml file in ORBSLAM3 and it works for custom video.