UZ-SLAMLab/ORB_SLAM3 #535: How to save and load map?

JACKLiuDay, 2022-03-31:
Hi, thank you for your great work about ORB_SLAM3 V1.0.!
I tried to read the calibration pdf carefully. And I want to save the map for localization with my quadruped robot. After  I enter the Atlas param in the .yaml profile. It did not work and showed segmentation fault. And I didn't see any map in the file. So how to save the map int ORB_SLAM3 v1.0. I need some help.
  
[image]

JACKLiuDay, 2022-03-31:
[image]
I tried the .yaml with mono-inertial again. It did not show segmentation fault like the stereo-inertial. And I saved the map successfully. But when I load the map.osa. It shows errors.

JACKLiuDay, 2022-04-13:
[image]
Sometimes when I try to shutdown system and save atlas map, it often segmentation fault without map saved.

mascomen4, 2022-06-20:
Hey. As far as I know Save and Load functionality is still a bit buggy. It was working for me on some sequences on the KITTI dataset, but for some of them not.  
 
But there is the thing I noticed: If you're using monocular mode, the map changes significantly during the processing, so I got a lot of errors with load and saving of the Atlas on KITTI dataset in this mode. Please try to use the mode which has a scalar metric: either **mono-inertial**, either **stereo** mode

hxlei0827, 2023-06-28:
> Hi, thank you for your great work about ORB_SLAM3 V1.0.! I tried to read the calibration pdf carefully. And I want to save the map for localization with my quadruped robot. After I enter the Atlas [...]

hi, may i know how did you find the .osa file? 
I add          System.LoadAtlasFromFile: "test.osa" in Asus.yaml, but I cannot find any .osa file after running.
Thanks in advance

wengchenyang1, 2023-07-12:
> hi, may i know how did you find the .osa file? I add System.LoadAtlasFromFile: "test.osa" in Asus.yaml, but I cannot find any .osa file after running. Thanks in advance

Sometimes the .osa file is not saved if segmentation fault occurs during saving. 

[This](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/443#issuecomment-1154814254) seems to be a solution.

kenhuang1964, 2023-11-26:
Hey @JACKLiuDay, did you end up figuring out how to load and save maps? I am trying to localize the camera within a saved map. Can you please help me?

JACKLiuDay, 2023-11-27:
> Hey @JACKLiuDay, did you end up figuring out how to load and save maps? I am trying to localize the camera within a saved map. Can you please help me?

Hi, sorry bro. I am using lidar SLAM for my project now. I did not use ORB SLAM3 for a long time.  And I tried to use the map system but got nothing.

closed 2023-11-27 by JACKLiuDay

ThakurSarveshGit, 2024-02-08:
> Hey @JACKLiuDay, did you end up figuring out how to load and save maps? I am trying to localize the camera within a saved map. Can you please help me?

Hi, to simply save a map uncomment the file save param in corresponding .YAML file:

For ex: EuRoC.yaml has these params:
# If the LoadFile doesn't exist, the system give a message and create a new Atlas from scratch
System.LoadAtlasFromFile: "Session_MH01Mono"

# The store file is created from the current session, if a file with the same name exists it is deleted
#System.SaveAtlasToFile: "Session_MH01Mono"

I ran evaluation_example.sh, first time with save param uncommented and next time with loadFile command uncommented and it was able to save and load file respectively.

kenhuang1964, 2024-02-08:
> Hi, to simply save a map uncomment the file save param in corresponding .YAML file:
>
> For ex: EuRoC.yaml has these params:
>
> ```
> # If the LoadFile doesn't exist, the system give a message and create a [...]

Thank you!

BeenToJupiter, 2024-03-16:
> Hi, to simply save a map uncomment the file save param in corresponding .YAML file:
>
> For ex: EuRoC.yaml has these params:
>
> ```
> # If the LoadFile doesn't exist, the system give a message and create a [...]

None of the .yaml files have any commented lines for that. What and where am I missing?

BeenToJupiter, 2024-03-16:
> Hi, to simply save a map uncomment the file save param in corresponding .YAML file:
>
> For ex: EuRoC.yaml has these params:
>
> ```
> # If the LoadFile doesn't exist, the system give a message and create a [...]

I added a file address for the save parameter, and it says that the binary files is saved to the given address but I am not able to find the file. What could be the issue? I do not see any seg fault.

tianyilim, 2025-09-27:
If anyone is still stuck on this in 2025 (maybe look at a better-implemented library? 😉) , this may help:

https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/882#issuecomment-3341706147