UZ-SLAMLab/ORB_SLAM3 #538: ./build_ros.sh error

hawas20, 2022-04-04:
Hi , i'm trying to install ORB-SLAM3 on Ubuntu 20.04
but i'm facing some error 
first i export path >>> 
 export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/ai/catkin_ws5/src/ORB_SLAM3/Examples/ROS
 but still have same error 
and i have 5 workspace 

./build_ros.sh
Building ROS nodes
mkdir: cannot create directory ‘build’: File exists
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:7 (cmake_policy):
  The OLD behavior for policy CMP0011 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:16 (cmake_policy):
  The OLD behavior for policy CMP0002 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:18 (cmake_policy):
  The OLD behavior for policy CMP0003 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:20 (cmake_policy):
  The OLD behavior for policy CMP0005 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:23 (cmake_policy):
  The OLD behavior for policy CMP0011 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

[rosbuild] Building package ORB_SLAM3
[rosbuild] Error from syntax check of ORB_SLAM3/manifest.xml
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/opt/ros/noetic/lib/python3/dist-packages/roslib/__init__.py", line 50, in <module>
    from roslib.launcher import load_manifest  # noqa: F401
  File "/opt/ros/noetic/lib/python3/dist-packages/roslib/launcher.py", line 42, in <module>
    import rospkg
ModuleNotFoundError: No module named 'rospkg'
CMake Error at /opt/ros/noetic/share/ros/core/rosbuild/private.cmake:77 (message):
  [rosbuild] Syntax check of ORB_SLAM3/manifest.xml failed; aborting
Call Stack (most recent call first):
  /opt/ros/noetic/share/ros/core/rosbuild/public.cmake:174 (_rosbuild_check_manifest)
  CMakeLists.txt:4 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/ai/catkin_ws5/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/build/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found.  Stop.

Jhonan01, 2022-04-21:
I'm taking this same error @hawas20

shubham300ska, 2022-05-25:
I also face same Problem any one have solution of this one please replay

Jhonan01, 2022-05-25:
> I also face same Problem any one have solution of this one please replay thank you

raincold23, 2022-06-08:
Same issue here, any suggestions?
Many thanks.

jj-kong, 2022-07-07:
yes, i have the same problem.

Abonaventure, 2022-07-14:
I have the same problem as follows:
[rosbuild] Building package ORB_SLAM3
[rosbuild] Error from syntax check of ORB_SLAM3/manifest.xml
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/opt/ros/noetic/lib/python3/dist-packages/roslib/__init__.py", line 53, in <module>
    import roslib.stacks  # noqa: F401, I100
  File "/opt/ros/noetic/lib/python3/dist-packages/roslib/stacks.py", line 45, in <module>
    import roslib.packages
  File "/opt/ros/noetic/lib/python3/dist-packages/roslib/packages.py", line 48, in <module>
    from catkin.find_in_workspaces import find_in_workspaces as catkin_find
  File "/opt/ros/noetic/lib/python3/dist-packages/catkin/find_in_workspaces.py", line 39, in <module>
    from catkin_pkg.packages import find_packages
  File "/opt/ros/noetic/lib/python3/dist-packages/catkin_pkg/packages.py", line 38, in <module>
    from .package import _get_package_xml
  File "/opt/ros/noetic/lib/python3/dist-packages/catkin_pkg/package.py", line 44, in <module>
    from catkin_pkg.condition import evaluate_condition
  File "/opt/ros/noetic/lib/python3/dist-packages/catkin_pkg/condition.py", line 17, in <module>
    import pyparsing as pp
  File "/opt/ros/noetic/lib/python3/dist-packages/pyparsing/__init__.py", line 100
    major: int
         ^
SyntaxError: invalid syntax
CMake Error at /opt/ros/noetic/share/ros/core/rosbuild/private.cmake:77 (message):
  [rosbuild] Syntax check of ORB_SLAM3/manifest.xml failed; aborting
Call Stack (most recent call first):
  /opt/ros/noetic/share/ros/core/rosbuild/public.cmake:174 (_rosbuild_check_manifest)
  CMakeLists.txt:4 (rosbuild_init)

-- Configuring incomplete, errors occurred!

ooswald216, 2022-08-29:
i also meet this questions any suggests?
thanks a lot

2447643603jzp, 2022-09-29:
python版本不对，改python软链接。按照说明进行操作。
cd /usr/bin
sudo rm python
sudo ln -s python3.8 python

2447643603jzp, 2022-09-29:
it will help

scut666, 2023-05-29:
rospack] Error: package 'roscpp' depends on non-existent package 'libboost-system-dev' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
[rospack] Error: package 'roscpp' depends on non-existent package 'libboost-chrono-dev' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
[rospack] Error: package 'roscpp' depends on non-existent package 'libboost-filesystem-dev' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
[rospack] Error: package 'roscpp' depends on non-existent package 'libboost-system-dev' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
[rospack] Error: package 'roslib' depends on non-existent package 'python-rospkg' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
[rospack] Error: package 'roslib' depends on non-existent package 'python3-rospkg' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
[rospack] Error: package 'cv_bridge' depends on non-existent package 'boost' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
[rospack] Error: package 'cv_bridge' depends on non-existent package 'libopencv-dev' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
[rospack] Error: package 'cv_bridge' depends on non-existent package 'python3' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
[rospack] Error: package 'cv_bridge' depends on non-existent package 'python3-opencv' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
[rospack] Error: package 'rosconsole' depends on non-existent package 'apr' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
[rospack] Error: package 'rosconsole' depends on non-existent package 'libboost-regex-dev' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
[rospack] Error: package 'rosconsole' depends on non-existent package 'log4cxx' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'

按照这样改之后就报了一堆找不到包的错