UZ-SLAMLab/ORB_SLAM3 #539: Size of matrix < 0 when loading rectification text when ros run Stereo_inertial

ShinChoon, 2022-04-04:
Hi I am running example bag file V1_02_medium.bag with 
rosrun ORB_SLAM3 Stereo_Inertial Vocabulary/ORBvoc.txt Examples_old/Stereo-Inertial/EuRoC.yaml true
It throw the error  in /home/ubuntu/opencv/modules/core/src/matrix.cpp:249 Asssertion failed for s>=0  in function: 'setSize'
-215 
the s value are rol and columns from left and right camera. 
the error happens when the final S value = -10234545663, but it doesn't exist in ymal file.

TioeAre, 2023-02-17:
Hello, I ran into a similar problem, have you reached a solution for this error?

ShinChoon, 2023-02-17:
> Hello, I ran into a similar problem, have you reached a solution for this error?

It was one year ago, I checed my note, there was no solution for that.

TioeAre, 2023-02-17:
ok, thank you very much.

ShinChoon, 2023-02-17:
> Hello, I ran into a similar problem, have you reached a solution for this error?

如果最后ORB SLAM3运行成功了，请告知我一下哈。我们去年跑测试的时候发现一旦保存并加载历史地图，程序就崩溃了。似乎因为电脑太弱。

TioeAre, 2023-02-17:
> > Hello, I ran into a similar problem, have you reached a solution for this error?
> 
> 如果最后ORB SLAM3运行成功了，请告知我一下哈。我们去年跑测试的时候发现一旦保存并加载历史地图，程序就崩溃了。似乎因为电脑太弱。

好的，我再试一试

TioeAre, 2023-02-18:
已解决，是由于cv_bridge默认的opencv为ros自带的4.2，编译orbslam3用的是我之前安装的4.5.5导致对不上。
解决办法：新建工作空间编译一份自己的cv_bridge。然后在orbslam3的ros例程中的cmakelists.txt用set设置cv_bridge到自己新编译的路径下。
参考连接：
https://github.com/opencv/opencv/issues/21367

https://blog.csdn.net/weixin_43868576/article/details/122688385

https://blog.csdn.net/u014015324/article/details/110850609

https://cyaninfinite.com/ros-cv-bridge-with-python-3/#Dependencies

https://blog.csdn.net/bigdog_1027/article/details/79092263

我的报错是这两种：
```bash
'cv::Exception' what():  OpenCV(4.5.5) /home/ubuntu/Downloads/opencv-4.5.5/opencv/modules/core/src/matrix.cpp:250: error: (-215:Assertion failed) s >= 0 in function 'setSize'`
'cv::Exception' what():  OpenCV(4.5.5) /home/ubuntu/Downloads/opencv-4.5.5/opencv/modules/core/src/alloc.cpp:73: error: (-4:Insufficient memory) Failed to allocate 87219830269440 bytes in function 'OutOfMemoryError
```