UZ-SLAMLab/ORB_SLAM3 #542: The clang compiler does not support '-march=native'

retoc71586, 2022-04-08:
Good evening everyone.
I am trying to install the project on my m1 Mac.
I managed to install all of the requirements but running ./build.sh it doesn't work.
I repetitively get the same error:  `the clang compiler does not support '-march=native'` which results in: ```
Undefined symbols for architecture x86_64:
  "_main", referenced from:
     implicit entry/start for main executable
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

retoc71586, 2022-04-09:
I managed to go a little further by substituting this flag in the CMakeList.txt file everywhere by: `-mcpu=apple-m1`.
However now when building I get 2 kind of errors: 
  1st of the kind `fatal error: 'something.h' file not found`
  2nd `Undefined symbols for architecture x86_64:
  "_main", referenced from:
     implicit entry/start for main executable
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
` or similarly `ld: warning: ignoring file something.o, building for macOS-x86_64 but attempting to link with file built for unknown-arm64
Undefined symbols for architecture x86_64:
`

retoc71586, 2022-04-09:
For the first kind of errors let me be more specific. From my understanding They come from the fact that clang++ compiler can not use libstdc++.
The first problem I had was with stdint-gcc.h  and I could solve it just by replacing it  with <stdint.h>
Then I had problems with  boost:	I have downloaded it trough home-brew but clang still couldn’t find it so I edited the CMakeList.txt files  to include /usr/local/include path: 
	set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS}  -I/opt/local/include -Wall  -O3 -mcpu=apple-m1 ")
	set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -I/opt/local/include -Wall  -O3 -mcpu=apple-m1")

Now I am having problems with tr1 which is not found. Any hint on this?

retoc71586, 2022-04-10:
I want to be a little more clear on the architecture errors I am getting because I narrowed them down to two occurrences:

1) The first time they appear is after [100%] Linking CXX shared library ../lib/libDBoW2.dylib, in Scanning dependencies of target DBoW2. They are all related to my opencv installation and raise errors of this kind:
`ld: warning: ignoring file /opt/anaconda3/envs/3D-vision-project/lib/libopencv_gapi.4.5.5.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64`
However I am pretty sure I have installed the arm64 packages, directly trough conda with the command `conda install -c conda-forge opencv`.

2) The second time when they occur is after `Build files have been written to: path_to_orbslam3/ORB_SLAM3/Thirdparty/Sophus/build`:

I still get the `ld: warning: ignoring file CMakeFiles/test_common.dir/test_common.cpp.o, building for macOS-x86_64 but attempting to link with file built for unknown-arm64
` kind of errors but here there is also a more serious issue:
`Undefined symbols for architecture x86_64:
  "_main", referenced from:
     implicit entry/start for main executable
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [test/core/test_common] Error 1
make[1]: *** [test/core/CMakeFiles/test_common.dir/all] Error 2
`

I hope I have been clear enough for someone to give me some help with the building procedure

retoc71586, 2022-04-10:
I have managed to get the Sophus build to work by downloading a different CMakeLists.txt file from the sops GitHub and modifying it a little. 
The only problem left is with the build of DBoW2. I do not know why but the linker looks for x86 libraries when I have downloaded the arm64 trough `conda install -c conda-forge opencv`

papasega, 2023-03-29:
Hello You can change : -march=native  by -mcpu=apple-m1

mccharley, 2023-05-04:
Have you tried `export SYSTEM_VERSION_COMPAT=1`? 
[used here to address a similar issue with the compiler](https://stackoverflow.com/a/65245427/8076929)

Worked for me.

Qucy, 2023-08-13:
> Have you tried `export SYSTEM_VERSION_COMPAT=1`? [used here to address a similar issue with the compiler](https://stackoverflow.com/a/65245427/8076929)

Thanks, this work for my Mac.