UZ-SLAMLab/ORB_SLAM3 #548: ORBSLAM3-v1, runros ORB_SLAM3 Mono,

h904798869, 2022-04-15:
hi，compeled successfully，runros ORB_SLAM3 Mono
ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular
Aborted (core dumped)
How to deal with it？

martin90102, 2022-04-25:
I have the same situation with U，do u have any process on  this issue？

h904798869, 2022-04-25:
是我这里ROS-neotic-cv_bridge的OpenCV版本的问题,我卸载了cv_bridge后,从github重新下载并编译安装cv_bridge问题就解决了,希望可以帮到你

joelloo, 2022-05-09:
There could be many possible reasons for a segmentation fault. In my case however, I found that the default CMakeLists settings for one component linked against OpenCV 4.5, while another linked against OpenCV 3.2. This caused a conflict, and at runtime the wrong OpenCV libraries were accessed, causing a segfault. 

Specifically, the top-level CMakeLists required OpenCV version > 4, and it found my 4.5 installation. Meanwhile, ROS required OpenCV 3.2 (along with DBoW, if I remember correctly). I downgraded the top-level CMakeLists to use OpenCV 3.2, and was able to run without segfaults.