UZ-SLAMLab/ORB_SLAM3 #549: make: *** [Makefile:84: all] Error 2

nargenziano, 2022-04-15:
I'm trying to install ORB-SLAM3 on Ubuntu 20.04 LTS (WSL on Windows 10) and I get the following errors when executing
`./build.sh`.
I have already tried many solutions proposed in previous issues (including downgrading Eigen to 3.2.10 and adding `#include <unistd.h>` in System.h) but I am stuck at the same point.

Does anybody have any hint?

Below is the error log:

/mnt/d/Documenti/ETH/3DV/Pangolin/components/pango_core/include/sigslot/signal.hpp:1180:65: error: ‘slots_reference’ was not declared in this scope
 1180 |         cow_copy_type<list_type, Lockable> ref = slots_reference();
      |                                                  ~~~~~~~~~~~~~~~^~
/mnt/d/Documenti/ETH/3DV/ORB_SLAM3/src/LoopClosing.cc: In member function ‘void ORB_SLAM3::LoopClosing::MergeLocal2()’:
/mnt/d/Documenti/ETH/3DV/ORB_SLAM3/src/LoopClosing.cc:1810:20: warning: use of an operand of type ‘bool’ in ‘operator++’ is deprecated [-Wdeprecated]
 1810 |         mnFullBAIdx++;
      |                    ^~
/mnt/d/Documenti/ETH/3DV/ORB_SLAM3/src/LoopClosing.cc:1788:9: warning: unused variable ‘numTemporalKFs’ [-Wunused-variable]
 1788 |     int numTemporalKFs = 11; //TODO (set by parameter): Temporal KFs in the local window if the map is inertial.
      |         ^~~~~~~~~~~~~~
/mnt/d/Documenti/ETH/3DV/ORB_SLAM3/src/LoopClosing.cc: In member function ‘void ORB_SLAM3::LoopClosing::SearchAndFuse(const KeyFrameAndPose&, std::vector<ORB_SLAM
3::MapPoint*>&)’:
/mnt/d/Documenti/ETH/3DV/ORB_SLAM3/src/LoopClosing.cc:2134:13: warning: unused variable ‘numFused’ [-Wunused-variable]
 2134 |         int numFused = matcher.Fuse(pKFi,Scw,vpMapPoints,4,vpReplacePoints);
      |             ^~~~~~~~
/mnt/d/Documenti/ETH/3DV/ORB_SLAM3/src/LoopClosing.cc: In member function ‘void ORB_SLAM3::LoopClosing::SearchAndFuse(const std::vector<ORB_SLAM3::KeyFrame*>&, st
d::vector<ORB_SLAM3::MapPoint*>&)’:
/mnt/d/Documenti/ETH/3DV/ORB_SLAM3/src/LoopClosing.cc:2162:9: warning: unused variable ‘total_replaces’ [-Wunused-variable]
 2162 |     int total_replaces = 0;
      |         ^~~~~~~~~~~~~~
In file included from /mnt/d/Documenti/ETH/3DV/ORB_SLAM3/src/Tracking.cc:20:
/mnt/d/Documenti/ETH/3DV/ORB_SLAM3/include/Tracking.h: In constructor ‘ORB_SLAM3::Tracking::Tracking(ORB_SLAM3::System*, ORB_SLAM3::ORBVocabulary*, ORB_SLAM3::Fra
meDrawer*, ORB_SLAM3::MapDrawer*, ORB_SLAM3::Atlas*, ORB_SLAM3::KeyFrameDatabase*, const string&, int, ORB_SLAM3::Settings*, const string&)’:
/mnt/d/Documenti/ETH/3DV/ORB_SLAM3/include/Tracking.h:286:10: warning: ‘ORB_SLAM3::Tracking::bStepByStep’ will be initialized after [-Wreorder]
  286 |     bool bStepByStep;
      |          ^~~~~~~~~~~
/mnt/d/Documenti/ETH/3DV/ORB_SLAM3/include/Tracking.h:284:18: warning:   ‘ORB_SLAM3::FrameDrawer* ORB_SLAM3::Tracking::mpFrameDrawer’ [-Wreorder]
  284 |     FrameDrawer* mpFrameDrawer;
      |                  ^~~~~~~~~~~~~
/mnt/d/Documenti/ETH/3DV/ORB_SLAM3/src/Tracking.cc:46:1: warning:   when initialized here [-Wreorder]
   46 | Tracking::Tracking(System *pSys, ORBVocabulary* pVoc, FrameDrawer *pFrameDrawer, MapDrawer *pMapDrawer, Atlas *pAtlas, KeyFrameDatabase* pKFDB, const stri
ng &strSettingPath, const int sensor, Settings* settings, const string &_nameSeq):
      | ^~~~~~~~
In file included from /mnt/d/Documenti/ETH/3DV/ORB_SLAM3/src/Tracking.cc:20:
/mnt/d/Documenti/ETH/3DV/ORB_SLAM3/include/Tracking.h:331:10: warning: ‘ORB_SLAM3::Tracking::mbCreatedMap’ will be initialized after [-Wreorder]
  331 |     bool mbCreatedMap;
      |          ^~~~~~~~~~~~
/mnt/d/Documenti/ETH/3DV/ORB_SLAM3/include/Tracking.h:327:18: warning:   ‘unsigned int ORB_SLAM3::Tracking::mnFirstFrameId’ [-Wreorder]
  327 |     unsigned int mnFirstFrameId;
      |                  ^~~~~~~~~~~~~~
/mnt/d/Documenti/ETH/3DV/ORB_SLAM3/src/Tracking.cc:46:1: warning:   when initialized here [-Wreorder]
   46 | Tracking::Tracking(System *pSys, ORBVocabulary* pVoc, FrameDrawer *pFrameDrawer, MapDrawer *pMapDrawer, Atlas *pAtlas, KeyFrameDatabase* pKFDB, const stri
ng &strSettingPath, const int sensor, Settings* settings, const string &_nameSeq):
      | ^~~~~~~~
In file included from /mnt/d/Documenti/ETH/3DV/ORB_SLAM3/src/Tracking.cc:20:
/mnt/d/Documenti/ETH/3DV/ORB_SLAM3/include/Tracking.h:354:33: warning: ‘ORB_SLAM3::Tracking::mpCamera2’ will be initialized after [-Wreorder]
  354 |     GeometricCamera* mpCamera, *mpCamera2;
      |                                 ^~~~~~~~~
/mnt/d/Documenti/ETH/3DV/ORB_SLAM3/include/Tracking.h:321:15: warning:   ‘ORB_SLAM3::KeyFrame* ORB_SLAM3::Tracking::mpLastKeyFrame’ [-Wreorder]
  321 |     KeyFrame* mpLastKeyFrame;
      |               ^~~~~~~~~~~~~~
/mnt/d/Documenti/ETH/3DV/ORB_SLAM3/src/Tracking.cc:46:1: warning:   when initialized here [-Wreorder]
   46 | Tracking::Tracking(System *pSys, ORBVocabulary* pVoc, FrameDrawer *pFrameDrawer, MapDrawer *pMapDrawer, Atlas *pAtlas, KeyFrameDatabase* pKFDB, const stri
ng &strSettingPath, const int sensor, Settings* settings, const string &_nameSeq):
      | ^~~~~~~~
/mnt/d/Documenti/ETH/3DV/ORB_SLAM3/src/Tracking.cc: In member function ‘void ORB_SLAM3::Tracking::Track()’:
/mnt/d/Documenti/ETH/3DV/ORB_SLAM3/src/Tracking.cc:2169:96: warning: comparison of integer expressions of different signedness: ‘long unsigned int’ and ‘int’ [-Ws
ign-compare]
 2169 |         if((mCurrentFrame.mnId<(mnLastRelocFrameId+mnFramesToResetIMU)) && (mCurrentFrame.mnId > mnFramesToResetIMU) &&
      |                                                                             ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
/mnt/d/Documenti/ETH/3DV/ORB_SLAM3/src/Tracking.cc: In member function ‘void ORB_SLAM3::Tracking::SearchLocalPoints()’:
/mnt/d/Documenti/ETH/3DV/ORB_SLAM3/src/Tracking.cc:3415:13: warning: unused variable ‘matches’ [-Wunused-variable]
[... 30 lines cut ...]
 1180 |         cow_copy_type<list_type, Lockable> ref = slots_reference();
      |                                                  ~~~~~~~~~~~~~~~^~
In file included from /mnt/d/Documenti/ETH/3DV/Pangolin/components/pango_core/include/pangolin/utils/signal_slot.h:3,
                 from /mnt/d/Documenti/ETH/3DV/Pangolin/components/pango_windowing/include/pangolin/windowing/window.h:35,
                 from /mnt/d/Documenti/ETH/3DV/Pangolin/components/pango_display/include/pangolin/display/display.h:34,
                 from /mnt/d/Documenti/ETH/3DV/Pangolin/components/pango_core/include/pangolin/pangolin.h:38,
                 from /mnt/d/Documenti/ETH/3DV/ORB_SLAM3/include/Map.h:27,
                 from /mnt/d/Documenti/ETH/3DV/ORB_SLAM3/include/KeyFrameDatabase.h:30,
                 from /mnt/d/Documenti/ETH/3DV/ORB_SLAM3/include/KeyFrame.h:29,
                 from /mnt/d/Documenti/ETH/3DV/ORB_SLAM3/include/MapPoint.h:23,
                 from /mnt/d/Documenti/ETH/3DV/ORB_SLAM3/include/FrameDrawer.h:24,
in::VarState::GetOrCreateVar(const T&, const pangolin::VarMeta&) [with T = bool]’
/mnt/d/Documenti/ETH/3DV/Pangolin/components/pango_vars/include/pangolin/var/var.h:83:50:   required from ‘pangolin::Var<T>::Var(const T&, const pangolin::VarMeta
&) [with T = bool]’
/mnt/d/Documenti/ETH/3DV/Pangolin/components/pango_vars/include/pangolin/var/var.h:88:35:   required from ‘pangolin::Var<T>::Var(const string&, const T&) [with T 
= bool; std::string = std::__cxx11::basic_string<char>]’
/mnt/d/Documenti/ETH/3DV/Pangolin/components/pango_display/include/pangolin/display/widgets.h:153:50:   required from here
/mnt/d/Documenti/ETH/3DV/Pangolin/components/pango_core/include/sigslot/signal.hpp:1180:65: error: ‘slots_reference’ was not declared in this scope
 1180 |         cow_copy_type<list_type, Lockable> ref = slots_reference();
      |                                                  ~~~~~~~~~~~~~~~^~
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:89: CMakeFiles/ORB_SLAM3.dir/src/LocalMapping.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:102: CMakeFiles/ORB_SLAM3.dir/src/LoopClosing.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:76: CMakeFiles/ORB_SLAM3.dir/src/Tracking.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:711: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

woowloop, 2022-04-18:
try to compile with c++14
Modify CMakeLists.txt
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")-->set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++14")

nargenziano, 2022-04-18:
> try to compile with c++14 Modify CMakeLists.txt set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")-->set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++14")

Thanks for your answer. I still get the same error though.

AlienPa, 2022-04-19:
https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/549#issuecomment-1101011250
Actually i think by changing C++11 to C++14 a C++14 may required for building codes and i searched to find how to install C++14 on ubuntu can help me with that ?
@woowloop

simutisernestas, 2022-04-20:
Add `set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++14")` after this block:
```
if(COMPILER_SUPPORTS_CXX11)
   set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
   add_definitions(-DCOMPILEDWITHC11)
   message(STATUS "Using flag -std=c++11.")
elseif(COMPILER_SUPPORTS_CXX0X)
   set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++0x")
   add_definitions(-DCOMPILEDWITHC0X)
   message(STATUS "Using flag -std=c++0x.")
else()
   message(FATAL_ERROR "The compiler ${CMAKE_CXX_COMPILER} has no C++11 support. Please use a different C++ compiler.")
endif()

nargenziano, 2022-04-20:
> Add `set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++14")` in `CMakeLists.txt` after this block:
> 
> ```
> if(COMPILER_SUPPORTS_CXX11)
>    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
>    add_definitions(-DCOMPILEDWITHC11)
>    message(STATUS "Using flag -std=c++11.")
> elseif(COMPILER_SUPPORTS_CXX0X)
>    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++0x")
>    add_definitions(-DCOMPILEDWITHC0X)
>    message(STATUS "Using flag -std=c++0x.")
> else()
>    message(FATAL_ERROR "The compiler ${CMAKE_CXX_COMPILER} has no C++11 support. Please use a different C++ compiler.")
> endif()
> ```

Just to make sure, you mean ORB_SLAM3/CMakeLists.txt, right? Because I keep getting the same error.

simutisernestas, 2022-04-20:
> Just to make sure, you mean ORB_SLAM3/CMakeLists.txt, right? Because I keep getting the same error.

Yes. Try deleting `build` folder and start over. It worked from the first try for me.

nargenziano, 2022-04-20:
> Yes. Try deleting `build` folder and start over. It worked from the first try for me.

I deleted `ORB_SLAM3/build` but the same happens again. Do you have any further advice?

simutisernestas, 2022-04-20:
> > > > Add `set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++14")` in `CMakeLists.txt` after this block:
> > > > ```
> > > > if(COMPILER_SUPPORTS_CXX11)
> > > >    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
> > > >    add_definitions(-DCOMPILEDWITHC11)
> > > >    message(STATUS "Using flag -std=c++11.")
> > > > elseif(COMPILER_SUPPORTS_CXX0X)
> > > >    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++0x")
> > > >    add_definitions(-DCOMPILEDWITHC0X)
> > > >    message(STATUS "Using flag -std=c++0x.")
> > > > else()
> > > >    message(FATAL_ERROR "The compiler ${CMAKE_CXX_COMPILER} has no C++11 support. Please use a different C++ compiler.")
> > > > endif()
> > > > ```
> > > 
> > > 
> > > Just to make sure, you mean ORB_SLAM3/CMakeLists.txt, right? Because I keep getting the same error.
> > 
> > 
> > Yes. Try deleting `build` folder and start over. It worked from the first try for me.
> 
> I deleted `ORB_SLAM3/build` and repeated the following `chmod +x build.sh` `./build.sh` But the same happens again. Do you have any further advice?

Seems like a problem with `Pangolia`. Did you run `make install` & `ldconfig` after building it?

nargenziano, 2022-04-21:
> Seems like a problem with `Pangolia`. Did you run `make install` & `ldconfig` after building it?

I tried to build everything over again running `make install` & `ldconfig` to make sure, but the same happens again.

woowloop, 2022-04-21:
> [#549 (comment)](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/549#issuecomment-1101011250) Actually i think by changing C++11 to C++14 a C++14 may required for building codes and i searched to find how to install C++14 on ubuntu and i didn't find something useful. can you help me with that ? @woowloop

No installation c++14 required  in ubuntu (The new system already exists c++14)

jonah-chen, 2022-05-20:
I was encountering similar issues. Not touching any CMakeLists but changing the last part of build.sh to
```bash
echo "Configuring and building ORB_SLAM3 ..."

mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_STANDARD=14
make -j4
```
works for me.

Glean-jcy, 2022-07-17:
Hi author, have you solved this problem?II also had the same problem！

Glean-jcy, 2022-07-18:
> Hi author, have you solved this problem?II also had the same problem！
The problem was solved after I installed the latest version of pangolin.

projectqduE, 2022-08-13:
> I was encountering similar issues. Not touching any CMakeLists but changing the last part of build.sh to
> 
> ```shell
> echo "Configuring and building ORB_SLAM3 ..."
> 
> mkdir build
> cd build
> cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_STANDARD=14
> make -j4
> ```
> 
> works for me.

Thanks for your method! I successfully solved the problem~

vipulgarg1998, 2022-08-23:
> try to compile with c++14 Modify CMakeLists.txt set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")-->set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++14")

Thanks. It worked for me :)