UZ-SLAMLab/ORB_SLAM3 #552: ROS 2 option?

jnauticus, 2022-04-21:
Is there any ros2 implementation that exists? Or a plan for one? (Foxy)

davidlai0522, 2022-07-14:
I would like to ask if there is any update for ROS2 implementation? I am also planning to use it on ROS2 Foxy.

samuk, 2022-07-21:
+1 I'm on Humble.

I see there is a ROS1 version https://github.com/plusk01/orbslam3_ros

silviazpi, 2022-10-17:
Any update here?

Nikolaiarn, 2024-01-28:
@silviazpi @samuk @davidlai0522 @jnauticus Did any of you fins a working ROS2 wrapper for ORB_SLAM3 that includes mono_inertial? I might be a bit late, but any help would be appreciated.

TSoli, 2024-02-23:
> @silviazpi @samuk @davidlai0522 @jnauticus Did any of you fins a working ROS2 wrapper for ORB_SLAM3 that includes mono_inertial? I might be a bit late, but any help would be appreciated.

I am not sure but it seems there is a discussion about integrating it into openvslam [here](https://github.com/stella-cv/stella_vslam/issues/235).