UZ-SLAMLab/ORB_SLAM3 #553: ./build.sh Error

tan-may16, 2022-04-21:
am trying to build the package with the given set of instructions.
When using ./build.sh I get the following error

`Configuring and building Thirdparty/DBoW2 ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tanmay/ORB_SLAM3/Thirdparty/DBoW2/build
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: cannot create directory ‘build’: File exists
-- BUILD TYPE:Release
-- Compiling on Unix
CMake Warning (dev) at /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:310 (if):
if given arguments:

"TRUE"
An argument named "TRUE" appears in a conditional statement. Policy
CMP0012 is not set: if() recognizes numbers and boolean constants. Run
"cmake --help-policy CMP0012" for policy details. Use the cmake_policy
command to set the policy and suppress this warning.
Call Stack (most recent call first):
/usr/share/cmake-3.10/Modules/FindOpenMP.cmake:425 (_OPENMP_GET_SPEC_DATE)
CMakeLists.txt:47 (FIND_PACKAGE)
This warning is for project developers. Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:310 (if):
if given arguments:

"TRUE"
An argument named "TRUE" appears in a conditional statement. Policy
CMP0012 is not set: if() recognizes numbers and boolean constants. Run
"cmake --help-policy CMP0012" for policy details. Use the cmake_policy
command to set the policy and suppress this warning.
Call Stack (most recent call first):
/usr/share/cmake-3.10/Modules/FindOpenMP.cmake:425 (_OPENMP_GET_SPEC_DATE)
CMakeLists.txt:47 (FIND_PACKAGE)
This warning is for project developers. Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/tanmay/ORB_SLAM3/Thirdparty/g2o/build
[100%] Built target g2o
Configuring and building Thirdparty/Sophus ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tanmay/ORB_SLAM3/Thirdparty/Sophus/build
[ 16%] Built target test_sim2
[ 16%] Built target test_common
[ 25%] Built target test_se2
[ 45%] Built target test_so3
[ 50%] Built target test_rxso2
[ 50%] Built target test_so2
[ 58%] Built target HelloSO3
[ 66%] Built target test_se3
[ 75%] Built target test_velocities
[ 83%] Built target test_rxso3
[100%] Built target test_geometry
[100%] Built target test_sim3
Uncompress vocabulary ...
Configuring and building ORB_SLAM3 ...
/home/tanmay/ORB_SLAM3
Build type: Release
-- Using flag -std=c++11.
OPENCV VERSION:
4.5.5
-- BUILD TYPE:Release
-- Compiling on Unix
CMake Warning (dev) at /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:310 (if):
if given arguments:

"TRUE"
An argument named "TRUE" appears in a conditional statement. Policy
CMP0012 is not set: if() recognizes numbers and boolean constants. Run
"cmake --help-policy CMP0012" for policy details. Use the cmake_policy
command to set the policy and suppress this warning.
Call Stack (most recent call first):
/usr/share/cmake-3.10/Modules/FindOpenMP.cmake:425 (_OPENMP_GET_SPEC_DATE)
Thirdparty/g2o/CMakeLists.txt:47 (FIND_PACKAGE)
This warning is for project developers. Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:310 (if):
if given arguments:

"TRUE"
An argument named "TRUE" appears in a conditional statement. Policy
CMP0012 is not set: if() recognizes numbers and boolean constants. Run
"cmake --help-policy CMP0012" for policy details. Use the cmake_policy
command to set the policy and suppress this warning.
Call Stack (most recent call first):
/usr/share/cmake-3.10/Modules/FindOpenMP.cmake:425 (_OPENMP_GET_SPEC_DATE)
Thirdparty/g2o/CMakeLists.txt:47 (FIND_PACKAGE)
This warning is for project developers. Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/tanmay/ORB_SLAM3
make: *** No targets specified and no makefile found. Stop.
`

The issue is probably with the last part in build.sh-
mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Release
make -j4

Please let me know how it can be resolved. Thank you

JACKLiuDay, 2022-04-22:
It looks like your build file was written to ORB_SLAM3, not build folder.

tan-may16, 2022-04-22:
@JACKLiuDay I am still learning ubuntu and therefore not able to understand the error properly. 

I cloned the repository, installed all the dependencies and executed the following commands as mentioned in the readme file:

cd ORB_SLAM3
chmod +x build.sh
./build.sh

Could you please elaborate and help me with resolving the issue?
Thank you.

JACKLiuDay, 2022-04-22:
> make: *** No targets specified and no makefile found. Stop.

[image]
Hi, guy, look at this. It seems like your makefile made by cmake is not in build folder. May it is caused by command "cmake .." in the ORB_SLAM3 path. Not in build folder. The right way is to "cmake .." in the build folder. 
My opinion may not be correct. You can have some try.