UZ-SLAMLab/ORB_SLAM3 #562: when i try to excute EuRoc dataset, i can't see any viewer

taehoon11, 2022-05-02:
I tried to excute Euroc dataset, It wasn't work. (I can't get any viewer (map viewer & current frame))
so i need help....  ~_~

<my course download dataset (MH01)>
$ cd ~
$ mkdir -p Datasets/EuRoc
$ cd Datasets/EuRoc/
$ wget -c http://robotics.ethz.ch/~asl-datasets/ijrr_euroc_mav_dataset/machine_hall/MH_01_easy/MH_01_easy.zip
$ mkdir MH01
$ unzip MH_01_easy.zip -d MH01/ 

<my command (excute for mono)>
$ cd ~
$ cd ORB_SLAM3
$ ./Examples/Monocular/mono_euroc ./Vocabulary/ORBvoc.txt ./Examples/Monocular/EuRoC.yaml ~/Datasets/EuRoc/MH01 ./Examples/Monocular/EuRoC_TimeStamps/MH01.txt dataset-MH01_mono

< Execution result (Terminal) >
num_seq = 1
file name: dataset-MH01_mono
Loading images for sequence 0...LOADED!

-------

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular
Loading settings from ./Examples/Monocular/EuRoC.yaml
Camera1.k3 optional parameter does not exist...
	-Loaded camera 1
	-Loaded image info
	-Loaded ORB settings
Viewer.imageViewScale optional parameter does not exist...
	-Loaded viewer settings
System.LoadAtlasFromFile optional parameter does not exist...
System.SaveAtlasToFile optional parameter does not exist...
	-Loaded Atlas settings
System.thFarPoints optional parameter does not exist...
	-Loaded misc parameters
----------------------------------
SLAM settings: 
	-Camera 1 parameters (Pinhole): [ 458.65399169921875 457.29598999023438 367.21499633789062 248.375 ]
	-Camera 1 distortion parameters: [  -0.28340810537338257 0.073959067463874817 0.00019359000725671649 1.7618711353861727e-05 ]
	-Original image size: [ 752 , 480 ]
	-Current image size: [ 600 , 350 ]
	-Camera 1 parameters after resize: [  365.94735717773438 333.44500732421875 292.99069213867188 181.10678100585938 ]
	-Sequence FPS: 20
	-Features per image: 1000
	-ORB scale factor: 1.2000000476837158
	-ORB number of scales: 8
	-Initial FAST threshold: 20
	-Min FAST threshold: 7

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
There are 1 cameras in the atlas
Camera 0 is pinhole
First KF:0; Map init KF:0
New Map created with 346 points

_______________________________________________________________________________________________________

well... i launched ORB SLAM3 pkg, and I built mt pkg. Using ./build.sh and ./build_ros.sh 
when i type commands to my terminal,

Terminal1 : $ roscore
Terminal2 : $ rosrun ORB_SLAM3 Mono ~/ORB_SLAM3/Vocabulary/ORBvoc.txt ~/ORB_SLAM3/Examples/Monocular/RealSense_D435i.yaml /camera/image_raw:=/camera/color/image_raw /imu:=/camera/imu

Two screens are created. One is map viewer, the other one is current frame. and is works(i can see image & see map, keyframe..etc)

But Whenever i try to excute dataset, i can't see any viewer... obviously i always fail.....
so i neeeeeeeeeed any helllllllllllllllllllp....

vanshilshah97, 2022-05-02:
around line 83  change the last argument of the function to true ORB_SLAM3::System SLAM(argv[1],argv[2],ORB_SLAM3::System::MONOCULAR, true); which is false by default and then rebuild the package.

taehoon11, 2022-05-02:
OHHHHHHH I SUCCEED~!!!! Thanks U for your help....

-_- >>> :)

tungtungyan, 2022-06-01:
> in mono_euroc.cc around line 83 change the last argument of the function to true ORB_SLAM3::System SLAM(argv[1],argv[2],ORB_SLAM3::System::MONOCULAR, true); which is false by default and then rebuild the package.

I have the same problem but I did solve it with this method.

scteam1994, 2025-04-02:
> in mono_euroc.cc around line 83 change the last argument of the function to true ORB_SLAM3::System SLAM(argv[1],argv[2],ORB_SLAM3::System::MONOCULAR, true); which is false by default and then rebuild the package.

How did you know that? Where can I find the relevant documentation?  Thank you anyway.