UZ-SLAMLab/ORB_SLAM3 #566: Report an error and solving method

Hao-Starrr, 2022-05-06:
This error may occur on Ubuntu 20.04 and 18.04.

When running `./build.sh`, I got 
```
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:76: CMakeFiles/ORB_SLAM3.dir/src/Tracking.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:711: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```
The problem can be solved by add `add_compile_options(-std=c++14)` before CMakeLists.txt file.
Then it build successfully.

Same solving method for the error:
```
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:375：CMakeFiles/ORB_SLAM3.dir/src/MLPnPsolver.cpp.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:76：CMakeFiles/ORB_SLAM3.dir/src/Tracking.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:245：CMakeFiles/ORB_SLAM3.dir/src/Frame.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:102：CMakeFiles/ORB_SLAM3.dir/src/LoopClosing.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:323：CMakeFiles/ORB_SLAM3.dir/src/G2oTypes.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:232：CMakeFiles/ORB_SLAM3.dir/src/Optimizer.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:390：CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:84：all] Error 2
```

dframirez-usmc, 2022-05-16:
Worked for me. 
Solved other issue during /`build.sh` or `g2o` and `Sophus`.
Thank you.

anastaga, 2022-06-10:
Where exactly should i paste add_compile_options(-std=c++14) in order to make it work?

Hao-Starrr, 2022-06-11:
> Where exactly should i paste add_compile_options(-std=c++14) in order to make it work?

At the first line of CMakeLists.txt file.

CruxDevStuff, 2022-07-05:
Worked for me too, Thanks :)

MadanGTiwary, 2023-05-26:
Worked for me, thanks.

kenhuang1964, 2023-11-15:
Anyone getting this error?

make[1]: *** [CMakeFiles/Makefile2:148: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

jeremy7703, 2023-12-05:
> Anyone getting this error?
> 
> make[1]: *** [CMakeFiles/Makefile2:148: CMakeFiles/ORB_SLAM3.dir/all] Error 2 make: *** [Makefile:91: all] Error 2

I have found the same problem with my computer, have you solved this?

kenhuang1964, 2023-12-05:
> > Anyone getting this error?
> > make[1]: *** [CMakeFiles/Makefile2:148: CMakeFiles/ORB_SLAM3.dir/all] Error 2 make: *** [Makefile:91: all] Error 2
> 
> I have found the same problem with my computer, have you solved this?

Yes, I changed some settings to C++ 14 I think. It's been a while. I forgot lol

alanthomas29, 2024-05-29:
I had similar issues and updated the cmake file and i could see that after making the changes it builds fine. I will attach a copy my cmakelist.txt file
[CMakeLists.txt](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/15486037/CMakeLists.txt)

jaticona, 2025-05-18:
to enforce -std=c++14, you should be add following lines in ORB-SLAM3/CMAkeLists.txt
set(CMAKE_CXX_STANDARD 14)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_EXTENSIONS OFF)
add the lines befor "# Check C++11 or C++0x support"