UZ-SLAMLab/ORB_SLAM3 #569: Points and frames initialized quickly (ROS command)

LimSsun, 2022-05-12:
Hi, I tried to run ORB SLAM3 with using ROS.
When I run ORB SLAM3 with this command to use Mono-inertial version
"rosrun ORB_SLAM3 Mono_Inertial ~/ORB_SLAM3/Vocabulary/ORBvoc.txt ~/ORB_SLAM3/Examples/Monocular-Inertial/RealSense_D435i.yaml /camera/image_raw:=/camera/color/image_raw /imu:=/camera/imu", 
Screens, points, and frames appear, but points and frames are quickly initialized and disappear. It's happening and disappearing repeatedly. Below is an error message that appears in the terminal window when the command is entered.
Can I know the reason and solution for this error?

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular-Inertial
Loading settings from /home/hi-slam/ORB_SLAM3/Examples/Monocular-Inertial/RealSense_D435i.yaml
Camera1.k3 optional parameter does not exist...
	-Loaded camera 1
Camera.newHeight optional parameter does not exist...
Camera.newWidth optional parameter does not exist...
	-Loaded image info
	-Loaded IMU calibration
	-Loaded ORB settings
Viewer.imageViewScale optional parameter does not exist...
	-Loaded viewer settings
System.LoadAtlasFromFile optional parameter does not exist...
System.SaveAtlasToFile optional parameter does not exist...
	-Loaded Atlas settings
System.thFarPoints optional parameter does not exist...
	-Loaded misc parameters
----------------------------------
SLAM settings: 
	-Camera 1 parameters (Pinhole): [ 382.613 382.613 320.183 236.455 ]
	-Camera 1 distortion parameters: [  0 0 0 0 ]
	-Original image size: [ 640 , 480 ]
	-Current image size: [ 640 , 480 ]
	-Sequence FPS: 30
	-Gyro noise: 0.001
	-Accelerometer noise: 0.01
	-Gyro walk: 1e-06
	-Accelerometer walk: 0.0001
	-IMU frequency: 200
	-Features per image: 1250
	-ORB scale factor: 1.2
	-ORB number of scales: 8
	-Initial FAST threshold: 20
	-Min FAST threshold: 7

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
There are 1 cameras in the atlas
Camera 0 is pinhole
Starting the Viewer
First KF:0; Map init KF:0
New Map created with 182 points
Not enough motion for initializing. Reseting...
TRACK: Reset map because local mapper set the bad imu flag 
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 282
mnInitialFrameId = 0
72 Frames set to lost
First KF:12; Map init KF:0
New Map created with 216 points
Not enough motion for initializing. Reseting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
Not enough motion for initializing. Reseting...
TRACK: Reset map because local mapper set the bad imu flag 
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 514
mnInitialFrameId = 354
142 Frames set to lost

sandstorm666, 2022-05-13:
Can anyone solve this problem?

Fisine, 2022-10-31:
I meet the same problems.angyone solve?

EdogawaKazuki, 2022-11-04:
Calibrate the camera.

silviazpi, 2022-11-07:
> Calibrate the camera.

Already did and still no good. the map shows for 3-4 seconds and then restarts for no apparent reason.

silviazpi, 2022-11-11:
> > Calibrate the camera.
> 
> Already did and still no good. the map shows for 3-4 seconds and then restarts for no apparent reason.

Update: image size was too big. Rescaled it to 640x400 px and everything works just fine :)