UZ-SLAMLab/ORB_SLAM3 #571: Running  TUM1.yaml with ros successfully , but only appear map viewer and not appear current frame.

paGto, 2022-05-14:
ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular
Loading settings from /home/lididas/ORB_SLAM3/Examples/Monocular/TUM1.yaml
	-Loaded camera 1
Camera.newHeight optional parameter does not exist...
Camera.newWidth optional parameter does not exist...
	-Loaded image info
	-Loaded ORB settings
Viewer.imageViewScale optional parameter does not exist...
	-Loaded viewer settings
System.LoadAtlasFromFile optional parameter does not exist...
System.SaveAtlasToFile optional parameter does not exist...
	-Loaded Atlas settings
System.thFarPoints optional parameter does not exist...
	-Loaded misc parameters
----------------------------------
SLAM settings: 
	-Camera 1 parameters (Pinhole): [ 517.306 516.469 318.643 255.314 ]
	-Camera 1 distortion parameters: [  0.262383 -0.953104 -0.005358 0.002628 1.16331 ]
	-Original image size: [ 640 , 480 ]
	-Current image size: [ 640 , 480 ]
	-Sequence FPS: 30
	-Features per image: 1000
	-ORB scale factor: 1.2
	-ORB number of scales: 8
	-Initial FAST threshold: 20
	-Min FAST threshold: 7

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
There are 1 cameras in the atlas
Camera 0 is pinhole
Starting the Viewer
Local Mapping STOP
First KF:0; Map init KF:0
New Map created with 164 points
^CShutdown

Saving keyframe trajectory to KeyFrameTrajectory.txt ...
段错误 (核心已转储)

fayu-ma, 2022-05-17:
I have the same problem. Have you successed.
我遇到了同样的问题，请问你解决了吗

paGto, 2022-05-17:
opencv4.2

------------------&nbsp;原始邮件&nbsp;------------------
发件人: ***@***.***&gt;; 
发送时间: 2022年5月17日(星期二) 下午4:43
收件人: ***@***.***&gt;; 
抄送: ***@***.***&gt;; ***@***.***&gt;; 
主题: Re: [UZ-SLAMLab/ORB_SLAM3] Running  TUM1.yaml with ros successfully , but only appear map viewer and not appear current frame. (Issue #571)

我已经解决了它！我改变了 opencv 的版本/orb_slam3/Examples/ROS/orb_slam3/cmakelists.txt

opencv 是什么版本

—
直接回复此邮件，在 GitHub 上查看或取消订阅.
你正在接收这个，因为你创作这个线程。留言 ID ***@***.*** have solved it! I changed the version of opencv in /orb_slam3/Examples/ROS/orb_slam3/cmakelists.txt.what`s the version of opencv

fayu-ma, 2022-05-17:
你用orbslam3跑过自己采的数据吗

paGto, 2022-05-17:
没有