UZ-SLAMLab/ORB_SLAM3 #578: Is there a way to see ORB slam3 on Rviz?

LimSsun, 2022-05-24:
When ORB slam3 is run, the map viewer window appears and you can see it on the screen, but when you try to see ORB slam3 on Rviz, it seems that a global staus error appears due to an error in the imu value.

macTracyHuang, 2022-05-24:
> When ORB slam3 is run, the map viewer window appears and you can see it on the screen, but when you try to see ORB slam3 on Rviz, it seems that a global staus error appears due to an error in the imu value.

How do you implement visualization on RVIZ?

koksyuen, 2022-05-24:
> > When ORB slam3 is run, the map viewer window appears and you can see it on the screen, but when you try to see ORB slam3 on Rviz, it seems that a global staus error appears due to an error in the imu value.
> 
> How do you implement visualization on RVIZ?

https://github.com/thien94/orb_slam3_ros_wrapper