UZ-SLAMLab/ORB_SLAM3 #585: Seg Fault when starting ORB_SLAM3 with stereo_kitti

Waterfly222, 2022-05-31:
I'm trying to run ORB_SLAM3 on the Kitti benchmark (ideally using inertial too) but I'm  facing a seg fault while launching stereo_kitti at the start of it (right where the 2nd camera should load) and don't really understand what I can do to change that.

I tried to find the origin of the seg fault using gdb and it seems to be in the libORB_SLAM3.so library

```
[ all gbd starting stuff and credentials ]

Input sensor was set to: Stereo
Loading settings from KITTI00-02.yaml
    - Loaded camera 1
    - Loaded camera 2
Camera.newHeight optional parameter does not exist...
Camera.newWidth optional parameter does not exist...
    - Loaded image info
    - Loaded ORB setings
Viewer.imageViewScale optional parameter does not exist...
    - Loaded viewer settings
System.LoadAtlasFromFile optionnal parameter does not exist...
System.SaveAtlasToFile optionnal parameter does not exist...
    - Loaded Alas setings
System.thFarPoints optionnal parameter does not exist...
    - Loaded misc parameters
-----------------------------------------------
SLAM settings :
    - Camera 1 parameters (Pinhole) : [ 718.856 718.856 607.193 183.216 ]

program recieved signal SIGSEGV, Segmentation fault
0x00007ffff7b5bbe in ORB_SLAM3::operator<<(std::ostream&, ORB_SLAM3::Settings const&) () from /home/me/code_dir/ORB_SLAM3/libORB_SLAM3.so
(gbd)
```

Any idea to help for this ?
I've read some of the previous issues and didn't find any for a seg fault at the beginning of a run

I'm in a VM under Ubuntu 18.04 and without ROS installed (maybe it could be an issue...) with the 1.0 version of ORB_SLAM3 and I'm using the odometry 00 dataset from Kitti (just changed to load the images from 0 and 1 to 2 and 3 caus I only dl thoses, maybe it could be an issue?) and already succesfully launched the Mono version of it (and didn't have any issues). I also already tried out the Stereo and Stereo-inertial of TUM-vi and didn't have this issue.

I'm a beginner in c++ so maybe the anwser is obvious and I don't see it

sun-lingyu, 2022-07-20:
same problem met.

Hiber980, 2022-10-22:
I find an old solution proposed by DreamWaterFound to solve the problem by modifying file KITTIxx.yaml.
https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/18

nesquik011, 2024-07-10:
it did not fix it with me @Hiber980  i am using the latest version of orb slam3