UZ-SLAMLab/ORB_SLAM3 #591: Segment Fault when do EuRoC example

raincold23, 2022-06-09:
Hi, 
I use ubuntu 20.04 and opencv version is 4.6.
`./Examples/Monocular/mono_euroc ./Vocabulary/ORBvoc.txt ./Examples/Monocular-Inertial/EuRoC.yaml /home/****/Desktop/SLAM/ORB_SLAM3/EuRoC/MH_04_difficult ./Examples/Monocular/EuRoC_TimeStamps/MH04.txt dataset-MH04_mono` 
Then I got segment fault. The gdb output is as follow:

`[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
num_seq = 1
file name: dataset-MH04_mono
Loading images for sequence 0...LOADED!

-------

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular
Loading settings from ./Examples/Monocular-Inertial/EuRoC.yaml
Camera1.k3 optional parameter does not exist...
	-Loaded camera 1
	-Loaded image info
	-Loaded ORB settings
Viewer.imageViewScale optional parameter does not exist...
	-Loaded viewer settings
System.LoadAtlasFromFile optional parameter does not exist...
System.SaveAtlasToFile optional parameter does not exist...
	-Loaded Atlas settings
System.thFarPoints optional parameter does not exist...
	-Loaded misc parameters
----------------------------------
SLAM settings: 
	-Camera 1 parameters (Pinhole): [ 458.65399169921875 457.29598999023438 367.21499633789062 248.375 ]
	-Camera 1 distortion parameters: [  -0.28340810537338257 0.073959067463874817 0.00019359000725671649 1.7618711353861727e-05 ]
	-Original image size: [ 752 , 480 ]
	-Current image size: [ 600 , 350 ]
	-Camera 1 parameters after resize: [  365.94735717773438 333.44500732421875 292.99069213867188 181.10678100585938 ]
	-Sequence FPS: 20
	-Features per image: 1000
	-ORB scale factor: 1.2000000476837158
	-ORB number of scales: 8
	-Initial FAST threshold: 20
	-Min FAST threshold: 7

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
There are 1 cameras in the atlas
Camera 0 is pinhole
[New Thread 0x7fffe0fc4700 (LWP 3489814)]
[New Thread 0x7fffe07c3700 (LWP 3489815)]
[New Thread 0x7fffdffc2700 (LWP 3489816)]
[New Thread 0x7fffdcbee700 (LWP 3489817)]
[New Thread 0x7fffd7fff700 (LWP 3489818)]
[New Thread 0x7fffd77fe700 (LWP 3489819)]
[New Thread 0x7fffd6ffd700 (LWP 3489820)]
[New Thread 0x7fffd67fc700 (LWP 3489821)]
[New Thread 0x7fffd5ffb700 (LWP 3489822)]
[New Thread 0x7fffd57fa700 (LWP 3489823)]
[New Thread 0x7fffd4ff9700 (LWP 3489824)]
[New Thread 0x7fffb7fff700 (LWP 3489825)]
[New Thread 0x7fffb77fe700 (LWP 3489826)]
[New Thread 0x7fffb6ffd700 (LWP 3489827)]
[New Thread 0x7fffb67fc700 (LWP 3489828)]
[New Thread 0x7fffb5ffb700 (LWP 3489829)]
[New Thread 0x7fffb57fa700 (LWP 3489830)]
[New Thread 0x7fffb4ff9700 (LWP 3489831)]
[New Thread 0x7fff93740700 (LWP 3489832)]
[New Thread 0x7fff92f3f700 (LWP 3489833)]
[Thread 0x7fff92f3f700 (LWP 3489833) exited]
[Thread 0x7fff93740700 (LWP 3489832) exited]
[New Thread 0x7fff92f3f700 (LWP 3489834)]
[New Thread 0x7fff93740700 (LWP 3489835)]
[New Thread 0x7fff9273e700 (LWP 3489840)]
[New Thread 0x7fff91f3d700 (LWP 3489841)]
[Thread 0x7fff91f3d700 (LWP 3489841) exited]
[Thread 0x7fff9273e700 (LWP 3489840) exited]
Starting the Viewer
First KF:0; Map init KF:0
New Map created with 182 points
--Type <RET> for more, q to quit, c to continue without paging--
Thread 1 "mono_euroc" received signal SIGSEGV, Segmentation fault.
__GI___libc_free (mem=0x131) at malloc.c:3102
3102	malloc.c: No such file or directory.
(gdb) where
#0  __GI___libc_free (mem=0x131) at malloc.c:3102
#1  0x00007ffff496287a in g2o::HyperGraph::clear() () at /home/****/Desktop/SLAM/ORB_SLAM3/Thirdparty/g2o/lib/libg2o.so
#2  0x00007ffff496bf54 in g2o::OptimizableGraph::~OptimizableGraph() () at /home/****/Desktop/SLAM/ORB_SLAM3/Thirdparty/g2o/lib/libg2o.so
#3  0x00007ffff7dd1d29 in ORB_SLAM3::Optimizer::BundleAdjustment(std::vector<ORB_SLAM3::KeyFrame*, std::allocator<ORB_SLAM3::KeyFrame*> > const&, std::vector<ORB_SLAM3::MapPoint*, std::allocator<ORB_SLAM3::MapPoint*> > const&, int, bool*, unsigned long, bool) () at /home/****/Desktop/SLAM/ORB_SLAM3/lib/libORB_SLAM3.so
#4  0x00007ffff7dd2ded in ORB_SLAM3::Optimizer::GlobalBundleAdjustemnt(ORB_SLAM3::Map*, int, bool*, unsigned long, bool) () at /home/****/Desktop/SLAM/ORB_SLAM3/lib/libORB_SLAM3.so
#5  0x00007ffff7d5d2a1 in ORB_SLAM3::Tracking::CreateInitialMapMonocular() () at /home/****/Desktop/SLAM/ORB_SLAM3/lib/libORB_SLAM3.so
#6  0x00007ffff7d5dca0 in ORB_SLAM3::Tracking::MonocularInitialization() () at /home/****/Desktop/SLAM/ORB_SLAM3/lib/libORB_SLAM3.so
#7  0x00007ffff7d5f281 in ORB_SLAM3::Tracking::Track() () at /home/****/Desktop/SLAM/ORB_SLAM3/lib/libORB_SLAM3.so
#8  0x00007ffff7d610a1 in ORB_SLAM3::Tracking::GrabImageMonocular(cv::Mat const&, double const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) ()
    at /home/****/Desktop/SLAM/ORB_SLAM3/lib/libORB_SLAM3.so
#9  0x00007ffff7d0bc26 in ORB_SLAM3::System::TrackMonocular(cv::Mat const&, double const&, std::vector<ORB_SLAM3::IMU::Point, std::allocator<ORB_SLAM3::IMU::Point> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) () at /home/****/Desktop/SLAM/ORB_SLAM3/lib/libORB_SLAM3.so
#10 0x0000555555559756 in main ()
`
Any suggestions?

chesily, 2022-08-20:
I got the same error when I tried to save the map

DavyJoker, 2024-02-01:
I changed the version of opencv from 4.6.0 to 4.4.0, then it works. maybe you can try

hplegend, 2024-04-21:
I encounter this problem too, so there are anyone else can provide solutions?

hplegend, 2024-04-21:
@

> I encounter this problem too, so there are anyone else can provide solutions?

I re-compile the program with opencv4.8, and stereo_euroc successfully launched.