UZ-SLAMLab/ORB_SLAM3 #594: error while loading shared libraries: libDBoW2.s

jing-house, 2022-06-13:
when i input this:
rosrun ORB_SLAM3  RGBD ./Vocabulary/ORBvoc.txt ./Examples/ROS/ORB_SLAM3/D415.yaml
--------------------------------------------------------------------------------------------------------------------------------------
/home/mrw/catkin_orb3/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/RGBD: error while loading shared libraries: libDBoW2.so: cannot open shared object file: No such file or directory

i dont't know the method to solve this problem

umbnich, 2023-02-07:
Hi, 
same problem. How to solve?

Miracleqjy, 2023-03-01:
Hi,
Same problem. Have u solved it?

IamShubhamGupto, 2023-10-25:
same problem

my logs when I do `./build.sh`

```
(game) shubham@Shubhams-MBP ORB_SLAM3 % ./build.sh
Configuring and building Thirdparty/DBoW2 ...
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is AppleClang 15.0.0.15000040
-- The CXX compiler identification is AppleClang 15.0.0.15000040
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done (0.4s)
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   DBoW2

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done (0.0s)
-- Build files have been written to: /Users/shubham/Documents/workspace/robotperception/project/ORB_SLAM3/Thirdparty/DBoW2/build
[ 28%] Building CXX object CMakeFiles/DBoW2.dir/DBoW2/BowVector.cpp.o
[ 57%] Building CXX object CMakeFiles/DBoW2.dir/DBoW2/ScoringObject.cpp.o
[ 71%] Building CXX object CMakeFiles/DBoW2.dir/DBoW2/FORB.cpp.o
[ 71%] Building CXX object CMakeFiles/DBoW2.dir/DBoW2/FeatureVector.cpp.o
[ 71%] Building CXX object CMakeFiles/DBoW2.dir/DUtils/Timestamp.cpp.o
[ 85%] Building CXX object CMakeFiles/DBoW2.dir/DUtils/Random.cpp.o
clang: warning: argument unused during compilation: '-L/Users/shubham/Downloads/boost_1_83_0/stage/lib' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-L/Users/shubham/Downloads/boost_1_83_0/stage/lib' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-L/Users/shubham/Downloads/boost_1_83_0/stage/lib' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-L/Users/shubham/Downloads/boost_1_83_0/stage/lib' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-L/Users/shubham/Downloads/boost_1_83_0/stage/lib' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-L/Users/shubham/Downloads/boost_1_83_0/stage/lib' [-Wunused-command-line-argument]
/Users/shubham/Documents/workspace/robotperception/project/ORB_SLAM3/Thirdparty/DBoW2/DUtils/Timestamp.cpp:99:2: warning: 'sprintf' is deprecated: This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead. [-Wdeprecated-declarations]
        sprintf(buf, "%.6lf", this->getFloatTime());
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/stdio.h:180:1: note: 'sprintf' has been explicitly marked deprecated here
__deprecated_msg("This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead.")
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/include/sys/cdefs.h:215:48: note: expanded from macro '__deprecated_msg'
        #define __deprecated_msg(_msg) __attribute__((__deprecated__(_msg)))
                                                      ^
1 warning generated.
[100%] Linking CXX shared library /Users/shubham/Documents/workspace/robotperception/project/ORB_SLAM3/Thirdparty/DBoW2/lib/libDBoW2.dylib
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: build: File exists
CMake Deprecation Warning at CMakeLists.txt:1 (CMAKE_MINIMUM_REQUIRED):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

[... 85 lines cut ...]
-- Compiling on Unix
-- Could NOT find OpenMP_C (missing: OpenMP_C_FLAGS OpenMP_C_LIB_NAMES)
-- Could NOT find OpenMP_CXX (missing: OpenMP_CXX_FLAGS OpenMP_CXX_LIB_NAMES)
-- Could NOT find OpenMP (missing: OpenMP_C_FOUND OpenMP_CXX_FOUND)
-- Configuring done (0.5s)
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   ORB_SLAM3
   g2o

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done (0.1s)
-- Build files have been written to: /Users/shubham/Documents/workspace/robotperception/project/ORB_SLAM3/build
[ 29%] Built target g2o
make[2]: *** No rule to make target `/Users/shubham/Documents/workspace/robotperception/project/ORB_SLAM3/Thirdparty/DBoW2/lib/libDBoW2.so', needed by `/Users/shubham/Documents/workspace/robotperception/project/ORB_SLAM3/lib/libORB_SLAM3.dylib'.  Stop.
make[1]: *** [CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [all] Error 2
(game) shubham@Shubhams-MBP ORB_SLAM3 %
```

TakShimoda, 2024-01-19:
I don't have that issue but did encounter it in other ROS packages. It's because your executable `RGBD.so` under `ROS/ORB_SLAM3` can't find `libDBoW2.so`. Go to the `ROS/ORB_SLAM3` folder and type: 

`ldd -r ./RGBD | grep DBoW`

It should show something like 

`libDBoW2.so => <your folder>/ORB_SLAM3/Thirdparty/DBoW2/lib/libDBoW2.so (0x00007feae2efc000)`. 

If it doesn't it means it can't find it or found DBoW2 elsewhere. I had this issue where it linked to DBoW2 from another ROS workspace because I had that workspace sourced in my bashrc when building the package. Make sure to only source setup.bash from `/opt/ros` and no other workspace and then build again (should probably delete the build folder inside the ROS workspace before building again).