UZ-SLAMLab/ORB_SLAM3 #596: the real size

Sherlock-hh, 2022-06-14:


Waterfly222, 2022-06-14:
In monocular you don't have the scale of you map because you don't have any depth knowledge so you'll never have your results in meters but on a arbitrary unit.
If you want to compare multiple results you need to rescale them to be at the same scale (I use the python package evo by Michael Grupp to do so)
I could be wrong though, I'm only a beginner in SLAM.

Sherlock-hh, 2022-06-21:
aha I find the better way to get the better MapPoints if there in a plane,  just change the construct way to:
`thread threadH(&TwoViewReconstruction::FindHomography, this, ref(vbMatchesInliersH), ref(SH), ref(H));`
 the Homography is better than Fundamental matrix if the features are in a plane

closed 2022-06-21 by Sherlock-hh

lcc157, 2023-03-10:
Do you know why it needs to be invMedianDepth = 4.0f/medianDepth , and this constant  (4.0f) is set according to what?

Sherlock-hh, 2023-03-10:
> 

it's means to set the average depth of mappoints to 4, because mono mode dosen't has the real scale of ture world, so u can set a average depth when the map is initialize, offer the map a scale first, it's been so long, i just remenber somewhere in code has set the mappoint'depth to 0~1, u can check the output of mappoint after initialized, the depth is around 4.

NJNischal, 2023-04-12:
Hello @Sherlock-hh , we’re you able to achieve real depth? Also, did you remove the scaling of the algorithm for your purpose? Thank you!