UZ-SLAMLab/ORB_SLAM3 #598: sophus error while running the mono-inertial

ricardo11235, 2022-06-16:
when i run the Monocular-Inertial Node,the following happened

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- Camera: Pinhole
- Image scale: 1
- fx: 458.654
- fy: 457.296
- cx: 367.215
- cy: 248.375
- k1: -0.283408
- k2: 0.0739591
- p1: 0.00019359
- p2: 1.76187e-05
- fps: 24
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Left camera to Imu Transform (Tbc): 
[3.4028235e+38, 0, 0, 0;
 0, 0, 0, 0;
 0, 0, 0, 0;
 0, 0, 0, 0]
Sophus ensure failed in function 'Sophus::SO3<Scalar_, Options>::SO3(const Transformation&) [with Scalar_ = float; int Options = 0; Sophus::SO3<Scalar_, Options>::Transformation = Eigen::Matrix<float, 3, 3>]', file '/home/bling/catkin_ws/src/ORB_SLAM3/Thirdparty/Sophus/sophus/so3.hpp', line 425.
R is not orthogonal:
 inf   0   0
  0   0   0
  0   0   0
已放弃 (核心已转储)

why does it happen and how to fix it?

Tao-11-chen, 2022-06-17:
same error

cy-2022, 2022-06-28:
For me it is the RGB-D-inertial case, and R is filled with nans...
Have been thinking if it is due to the calibration of the IMU and/or the joint-calibration of the IMU and the camera.

macTracyHuang, 2022-07-19:
> For me it is the RGB-D-inertial case, and R is filled with nans... Have been thinking if it is due to the calibration of the IMU and/or the joint-calibration of the IMU and the camera.

do u know if a joint-calibration of the IMU and the camera is necessary to run it? thx

yyanzzw, 2023-06-08:
me too, my case is stereo_inertial, have anyone been solved this problem?