UZ-SLAMLab/ORB_SLAM3 #600: localization mode accuracy with TUM dataset

macTracyHuang, 2022-06-21:
hello, I'm trying the localization only mode with the TUM dataset(rgbd_dataset_freiburg2_large_with_loop) and found the predicted pose is far from the trajectory that was saved by SLAM mode with exactly same dataset.

1.I first generated keyframetrajectory.txt and save map with the dataset in SLAM mode in .bin extension.
2.Load the map , turn on localization mode and predict pose information with the same dataset.

I was expecting exatly the same trajectory but found the pose is far from the save maps.

Could any one give me a hint what would possibily cause this problem?

Thx a lot

Vslamer, 2022-07-25:
I also encountered this problem ( ex: information that was loaded is not as completed as the one in SLAM mode), the loaded map points are few, did you solve this problem？

macTracyHuang, 2022-07-27:
> I also encountered this problem ( ex: information that was loaded is not as completed as the one in SLAM mode), the loaded map points are few, did you solve this problem？

I found out when localization mode is on, it will create new map everytime, somehow the active map will not be the one I loaded. I currently save the map with the biggest number of keyframes as my map. after that, instead of creating a new map, I set the map I loaded as the active map. This solution is ok for my current purpose but I believe it's not the best way to correct it. The modifed code is in my github fork repository, you can have look of it and welcome to discuss with me.