UZ-SLAMLab/ORB_SLAM3 #603: (Catkin_ws) in error ./build.sh

alstjr510, 2022-06-25:
(slam) jangminseock@me:~/jms_ws/src/ORB_SLAM3$ ./build.sh 
Configuring and building Thirdparty/DBoW2 ...
mkdir: `build' 디렉토리를 만들 수 없습니다: 파일이 있습니다
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jangminseock/jms_ws/src/ORB_SLAM3/Thirdparty/DBoW2/build
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: `build' 디렉토리를 만들 수 없습니다: 파일이 있습니다
-- BUILD TYPE:Release
-- Compiling on Unix
CMake Warning (dev) at /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:310 (if):
  if given arguments:

    "TRUE"

  An argument named "TRUE" appears in a conditional statement.  Policy
  CMP0012 is not set: if() recognizes numbers and boolean constants.  Run
  "cmake --help-policy CMP0012" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:425 (_OPENMP_GET_SPEC_DATE)
  CMakeLists.txt:47 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:310 (if):
  if given arguments:

    "TRUE"

  An argument named "TRUE" appears in a conditional statement.  Policy
  CMP0012 is not set: if() recognizes numbers and boolean constants.  Run
  "cmake --help-policy CMP0012" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:425 (_OPENMP_GET_SPEC_DATE)
  CMakeLists.txt:47 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/jangminseock/jms_ws/src/ORB_SLAM3/Thirdparty/g2o/build
[100%] Built target g2o
Configuring and building Thirdparty/Sophus ...
mkdir: `build' 디렉토리를 만들 수 없습니다: 파일이 있습니다
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jangminseock/jms_ws/src/ORB_SLAM3/Thirdparty/Sophus/build
[ 12%] Built target test_sim3
[ 16%] Built target test_common
[ 41%] Built target test_se2
[ 41%] Built target test_rxso2
[ 41%] Built target test_sim2
[ 50%] Built target test_so2
[ 62%] Built target HelloSO3
[ 66%] Built target test_geometry
[ 75%] Built target test_se3
[ 83%] Built target test_rxso3
[ 91%] Built target test_so3
[100%] Built target test_velocities
Uncompress vocabulary ...
Configuring and building ORB_SLAM3 ...
mkdir: `build' 디렉토리를 만들 수 없습니다: 파일이 있습니다
Build type: Release
-- Using flag -std=c++11.
OPENCV VERSION:
4.2.0
CMake Warning at CMakeLists.txt:43 (find_package):
  By not providing "Findrealsense2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "realsense2", but CMake did not find one.

  Could not find a package configuration file provided by "realsense2" with
  any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

  Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.

-- BUILD TYPE:Release
-- Compiling on Unix
CMake Warning (dev) at /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:310 (if):
  if given arguments:

    "TRUE"

  An argument named "TRUE" appears in a conditional statement.  Policy
  CMP0012 is not set: if() recognizes numbers and boolean constants.  Run
  "cmake --help-policy CMP0012" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:425 (_OPENMP_GET_SPEC_DATE)
  Thirdparty/g2o/CMakeLists.txt:47 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:310 (if):
  if given arguments:

    "TRUE"

  An argument named "TRUE" appears in a conditional statement.  Policy
  CMP0012 is not set: if() recognizes numbers and boolean constants.  Run
  "cmake --help-policy CMP0012" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:425 (_OPENMP_GET_SPEC_DATE)
  Thirdparty/g2o/CMakeLists.txt:47 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/jangminseock/jms_ws/src/ORB_SLAM3/build
[ 29%] Built target g2o
[ 55%] Built target ORB_SLAM3
[ 57%] Built target stereo_inertial_tum_vi_old
[ 59%] Built target mono_inertial_tum_vi_old
[ 61%] Built target mono_kitti_old
[ 62%] Built target stereo_inertial_euroc_old
[ 64%] Built target stereo_euroc
[ 66%] Built target stereo_euroc_old
[ 68%] Built target mono_tum
[ 70%] Built target mono_tum_vi_old
[ 72%] Built target mono_euroc_old
[ 74%] Built target mono_inertial_euroc
[ 76%] Built target stereo_kitti
[ 78%] Built target mono_inertial_tum_vi
[ 79%] Built target mono_inertial_euroc_old
[ 81%] Built target rgbd_tum
[ 83%] Built target mono_kitti
[ 85%] Built target mono_euroc
[ 87%] Built target stereo_kitti_old
[ 88%] Built target mono_tum_vi
[ 90%] Built target stereo_tum_vi_old
[ 92%] Built target stereo_tum_vi
[ 94%] Built target stereo_inertial_euroc
[ 96%] Built target rgbd_tum_old
[ 98%] Built target stereo_inertial_tum_vi
[100%] Built target mono_tum_old

_**Error syntax**_
CMake Warning at CMakeLists.txt:43 (find_package):
  By not providing "Findrealsense2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "realsense2", but CMake did not find one.

  Could not find a package configuration file provided by "realsense2" with
  any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

I tried to solve the problem in the link below, but it comes out like the error phrase above. Please tell me how to solve it
(https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/118#issuecomment-726648539)

closed 2022-06-26 by alstjr510

Amber016, 2022-07-06:
Hi， I also meet the same error with yours, but i find that u have closed the issue. I think that u have solved the issue. Can you tell me how to solve it?

alstjr510, 2022-07-07:
> Hi， I also meet the same error with yours, but i find that u have closed the issue. I think that u have solved the issue. Can you tell me how to solve it?

>Yes, I downloaded the package realsense2 to Google and solved it.