UZ-SLAMLab/ORB_SLAM3 #605: Can someone explain the Tracking::TrackReferenceKeyFrame() function?

mel-ctrl, 2022-06-27:
Hi, can someone explain how theTrackReferenceKeyframe function works? Especially why a BoW model is used for the matching? 
Why doesn't ORB-SLAM directly match the feature points in the reference keyframe to the current frame. Why is a BoW model needed (you already input the reference keyframe, so what does the BoW model retrieve?) and also what is the vpMapPointMatches input when calling the SearchByBoW model?

bool Tracking::TrackReferenceKeyFrame()
{
    // Compute Bag of Words vector
    mCurrentFrame.ComputeBoW();

    // We perform first an ORB matching with the reference keyframe
    // If enough matches are found we setup a PnP solver
    ORBmatcher matcher(0.9,true); //0.7
    vector<MapPoint*> vpMapPointMatches;

    **int nmatches = matcher.SearchByBoW(mpReferenceKF,mCurrentFrame,vpMapPointMatches);**

Stevoooos, 2022-07-02:
Hello metl-ctrl!
In the function "SearchByBoW", the bag of words is used in an unusual way : it is used to accelerate the matching between two images (no realtionship with image indexation).

The idea comes from the observation that comparing descriptors is expensive while comparing visual words very inexpensive since it implies a simple integer comparison.  **Under the assumption that "two descriptors with different visual word are far in the descriptor space"**, you can reject all the match candidates that don't imply two features sharing the same visual word.  
Then, you can first compare the visual words. If they are not equal, you consider that the descriptors are too much different to be a matched , then it i useless to compare the ORB descriptors. If the  visual words are equal,  you know that the featrues are not super far in the descriptor space, then you have to compare the ORB descriptors to decide if the match is valid or not.

The second advantage is that the BoW  (DBOW2::FeatureVector) provides an access to the features that is ordered with respect to the visual words.  Then you can efficiently identify the features of the two images who share the same visual words by iterating only once the BoW of each image.

In conclusion, it is just a way to accelerate the matching process by reducing the number of comparison of feature desciptor.  

However, remember **there is a tradeoff between the acceleration and the performance** of the matching that comes from the fact that **the assumption that "two descriptors with different visual word are far in the descriptor space" can be invalid**:

- If you are using visual words from a very deep level, you will accelerate a lot the matching (the probability of the feature have the same visual word is low)  but you will miss lot of correct match because the assumption is often wrong ( lot of features may be near in descriptor space but not sharing the same visual word).

- If you are using  visual words that are very shallow, you will miss very few or no correct matchs (the assumption will be almost all the time correct) but you will not accelerate that much the process since most of the  features will have the same visual word (eg. if you use the level of the root, there is only one visual word, then the visual word will not filter anything).

About "vpMapPointMatches ", it is not an input but an output. It will contain all the 3D points that were associated with a feature that was matched. I understand that it can be surprising to find such output in a function that seems to focus only on 2D feature matching. I guess it was done because it was convenient and for performance reasons... but it breaks a bit the "single responsability principle" of programming. 

I hope the explainations are clear. If not, let me know.

closed 2022-07-12 by mel-ctrl

Mechazo11, 2022-09-28:
@Stevoooos Thank you very much for the detailed explanation.