UZ-SLAMLab/ORB_SLAM3 #610: fail to track local map!

sadsasuke, 2022-07-05:
Hi,
I am using monocular version with my own Data, Has someone faced this issue and any ideas on how I can process images before feeding into ORBSLAM to not face this issue!
Thanks,

davidlai0522, 2022-07-22:
Hi, I encountered the same issue when running ORB SLAM3 RGB-D in ROS. I dont have a physical RGB-D camera so i didnt go through calibration before starting. 

Here is the dataset I am using: https://vision.in.tum.de/data/datasets/rgbd-dataset/download#freiburg2_pioneer_slam

Below are the two command I run in the terminal:
rosrun ORB_SLAM3 RGBD Vocabulary/ORBvoc.txt Examples/RGB-D/TUM1.yaml

rosbag play rgbd_dataset_freiburg2_pioneer_slam.bag /camera/rgb/image_color:=/camera/rgb/image_raw /camera/depth/image:=/camera/depth_registered/image_raw

Here is the output in the terminal:
[image]

Error:
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 0
mnInitialFrameId = 2186
126 Frames set to lost
First KF:219; Map init KF:216
New Map created with 932 points
Fail to track local map!
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 0
mnInitialFrameId = 2314
269 Frames set to lost
First KF:229; Map init KF:219
New Map created with 218 points
Fail to track local map!
Relocalized!!

macTracyHuang, 2022-08-11:
It literally means lost of tracking. You can observe your data to see when it failed, for example, pure rotation or a scene lacks enough matching point might cause the issue. Then you fix or avoid it when collecting data.

closed 2022-08-12 by sadsasuke

qn757275373, 2023-05-25:
Hi,i want to know what features of the image are more focused on by ORB feature points so that I can construct a suitable data set,thanks

Rajni-kanth, 2023-11-04:
what should i do if i have less features in my room like plain walls and plain corridors , I want to do it on realtime slam