UZ-SLAMLab/ORB_SLAM3 #612: Run "Examples/Monocular/mono_euroc" get "Segmentation fault (core dumped)"

gfjiyue, 2022-07-08:
num_seq = 1
Loading images for sequence 0...LOADED!

-------

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular
Loading settings from ./Examples/Monocular/EuRoC.yaml
Camera1.k3 optional parameter does not exist...
	-Loaded camera 1
	-Loaded image info
	-Loaded ORB settings
Viewer.imageViewScale optional parameter does not exist...
	-Loaded viewer settings
System.LoadAtlasFromFile optional parameter does not exist...
System.SaveAtlasToFile optional parameter does not exist...
	-Loaded Atlas settings
System.thFarPoints optional parameter does not exist...
	-Loaded misc parameters
----------------------------------
SLAM settings: 
	-Camera 1 parameters (Pinhole): [ 458.65399169921875 457.29598999023438 367.21499633789062 248.375 ]
	-Camera 1 distortion parameters: [  -0.28340810537338257 0.073959067463874817 0.00019359000725671649 1.7618711353861727e-05 ]
	-Original image size: [ 752 , 480 ]
	-Current image size: [ 600 , 350 ]
	-Camera 1 parameters after resize: [  365.94735717773438 333.44500732421875 292.99069213867188 181.10678100585938 ]
	-Sequence FPS: 20
	-Features per image: 1000
	-ORB scale factor: 1.2000000476837158
	-ORB number of scales: 8
	-Initial FAST threshold: 20
	-Min FAST threshold: 7

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
There are 1 cameras in the atlas
Camera 0 is pinhole
First KF:0; Map init KF:0
New Map created with 346 points
Segmentation fault (core dumped)

lacie-life, 2022-07-11:
You can try removing  native flag when building g2o and ORB_SLAM3. I met same problem with RGB-D example and removing fixed that.

Z-Jeff, 2022-08-29:
> example

I have removed ` -march=native ` in g2o and ORB_SLAM3, and rebuild.
But Segmentation fault still occurs.
I figure out  `SaveAtlas()` result in this error, but I have no idea how to solve this.

saadnec, 2022-10-18:
> You can try removing native flag when building g2o and ORB_SLAM3. I met same problem with RGB-D example and removing fixed that.

Could u please help me finding the native flag?

Gong2Kun, 2022-12-23:
> 

Have you solved it? I have the same problem. Can you help me?

JayChen1996, 2023-04-09:
You may need to reconstruct the data.csv, the format of "path_to_times_file" should be the name of the image, see [this line ](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/4452a3c4ab75b1cde34e5505a36ec3f9edcdc4c4/Examples/Monocular/mono_euroc.cc#L65) and [this line](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/4452a3c4ab75b1cde34e5505a36ec3f9edcdc4c4/Examples/Monocular/mono_euroc.cc#L221)

nagylilla96, 2023-06-11:
> > 
> 
> Have you solved it? I have the same problem. Can you help me?

In case anyone else has this issue, the flags mentioned by @lacie-life can be found in ORB-SLAM3/CMakeLists.txt lines 11-12 and ORB-SLAM3/Thirdparty/g2o/CMakeLists.txt lines 54-55.
This fixed the issue for me, but it would be nice if someone could explain why this was an issue in the first place!

haniraid, 2024-08-26:
after i uncommenetd the section as it says above, the procces dies just after images been proceed, with no seglentation fault ! i cant figure it out whats the issue. im using monocular camera and running 
./Examples/Monocular/mono_tum Vocabulary/ORBvoc.txt Examples/Monocular/TUM-VI.yaml /home/raidhani/ros1_ws/src/YOLO_ORB_SLAM3/resized_images
i resized images to 640*640 as the input for yolo, or thats what i thought it was right.
on the other half, whene i ran monocular with ros, slam works but without object detection
im using YOLO_ORB_SLAM3