UZ-SLAMLab/ORB_SLAM3 #613: I cannot show frame and map window on WSL

wooolwooolwoool, 2022-07-13:
I installed ORB_SLAM3 on WSL according to following page.
https://medium.com/@tristan.sch/setting-up-a-virtual-machine-with-orb-slam-3-1a12e7905cf5

After that, I tried to example code. It works fine on terminal, but The frame and map window did not appear. I can use GUI application, for example, gedit. I want show window. How to fix?

I show log.

```
$ ./Examples/Monocular/mono_euroc ./Vocabulary/ORBvoc.txt ./Examples/Monocular/EuRoC.yaml Datasets/EuRoc/MH01 ./Examples/Monocular/EuRoC_TimeStamps/MH01.txt dataset-MH01_mono
num_seq = 1
file name: dataset-MH01_mono
Loading images for sequence 0...LOADED!

-------

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular
Loading settings from ./Examples/Monocular/EuRoC.yaml
Camera1.k3 optional parameter does not exist...
        -Loaded camera 1
        -Loaded image info
        -Loaded ORB settings
Viewer.imageViewScale optional parameter does not exist...
        -Loaded viewer settings
System.LoadAtlasFromFile optional parameter does not exist...
System.SaveAtlasToFile optional parameter does not exist...
        -Loaded Atlas settings
System.thFarPoints optional parameter does not exist...
        -Loaded misc parameters
----------------------------------
SLAM settings:
        -Camera 1 parameters (Pinhole): [ 458.65399169921875 457.29598999023438 367.21499633789062 248.375 ]
        -Camera 1 distortion parameters: [  -0.28340810537338257 0.073959067463874817 0.00019359000725671649 1.7618711353861727e-05 ]
        -Original image size: [ 752 , 480 ]
        -Current image size: [ 600 , 350 ]
        -Camera 1 parameters after resize: [  365.94735717773438 333.44500732421875 292.99069213867188 181.10678100585938 ]
        -Sequence FPS: 20
        -Features per image: 1000
        -ORB scale factor: 1.2000000476837158
        -ORB number of scales: 8
        -Initial FAST threshold: 20
        -Min FAST threshold: 7

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name:
There are 1 cameras in the atlas
Camera 0 is pinhole
First KF:0; Map init KF:0
New Map created with 271 points
Shutdown

Saving trajectory to f_dataset-MH01_mono.txt ...
There are 1 maps in the atlas
  Map 0 has 313 KFs

End of saving trajectory to f_dataset-MH01_mono.txt ...

Saving keyframe trajectory to kf_dataset-MH01_mono.txt ...
```

gabriel-azeredo, 2022-11-09:
I am getting the exact same output and neither frame and map window as well. Were you able to resolve the issue?

gabriel-azeredo, 2022-11-09:
This issue was solved on #464