UZ-SLAMLab/ORB_SLAM3 #617: Error "waiting for images " Mono ROS

pie-leroy, 2022-07-18:
Hello,

I am blocked for several days, I can't run ORB SLAM3 Mono. I try to connect airsim (publisher) to ORB SLAM3, the code compiles and launches but does not emit any error. I don't receive the images but it seems to me that I have correctly configured the topic names "/camera/image_raw".

Here is an example of the image I can read with RVIZ:
[image]

The graph generated by rqt graph :
[image]

no received images:
[image]

I work on an ubuntu environment installed on windows (with X11 as graphical redirection)

Do you have an idea?
thank you in advance

ghkeller, 2022-07-20:
Hi -- I am experiencing the same thing. I know that the image is being published on the correct topic and being ingested by the callback (as I have included a CV window instance within the callback to present the cv_bridge image, which appears just fine), but it is stuck on "Waiting For Images". I also get the "Framebuffer with requested attributes..." line as seen in your screenshot. I am running on a dual-boot system on an Acer Nitro 5 with Ubuntu 18.04.

pie-leroy, 2022-07-20:
I have not been able to get this code to work, as an alternative solution for the moment I have managed to run this code : 
https://github.com/appliedAI-Initiative/orb_slam_2_ros
I
f ever this can unblock you

macTracyHuang, 2022-08-11:
Have you guys tried to wait longer to see if it works?

When I first run ORBSLAM, I expected  images show right after I play the bag and run the SLAM,  I terminated the process every time when I  see waiting for images. 
One day when I went to to toilet after commanding, when I came back, I found it starts to work!!

pie-leroy, 2022-08-25:
Hello,

I finally got the code to work. I modified in the "cpp" code the topic read in input and it worked.

mamdouhmostafa, 2023-03-02:
> Hello,
> 
> I finally got the code to work. I modified in the "cpp" code the topic read in input and it worked.

Could you explain more how did you do it?

uestyhox1, 2023-03-05:
> Hello,
> 
> I finally got the code to work. I modified in the "cpp" code the topic read in input and it worked.

follow asking

mamdouhmostafa, 2023-03-05:
> Hello,
> 
> I finally got the code to work. I modified in the "cpp" code the topic read in input and it worked.

If you could explain what did you modify in the CPP code and which file?
Thank you

julixit, 2024-01-22:
We used the Mono_Inertial method and also had no visual camera input. 
We then made changes in the file "ros_mono_inertial.cc" in "ORB_SLAM3/Examples_old/ROS"  
In line 99 we changed the camera source of the subscriber from "camera/image_raw" to "camera/left/image_raw" to fit the naming of the running rostopics. 
And now it works.