UZ-SLAMLab/ORB_SLAM3 #625: How to publish the velocity?

sijinsebastian, 2022-07-28:
There doesn't seems to be any topic available to get the velocity estimated by ORB_SLAM3 system. 

I am currently using Monocular mode using ROS and wanted to publish the velocity. Is it already available in some parameter or variable in the system. If someone knows this or if anyone was able to publish the velocity, it would be a great help for me if you could share some insights here. 
Thanks a lot in advance.

makolele12, 2022-08-17:
@sijinsebastian by default ORB-SLAM does not publish it. If you are in Monocular-Only you do not have a real scale and therefore you cannot calculate the velocity in real units. 

When using the IMU (in either Monocular or Stereo), there is a velocity Vector Vwb2:

https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/4452a3c4ab75b1cde34e5505a36ec3f9edcdc4c4/src/Tracking.cc#L1774

Here you have velocity in three different axes. You should then decide in which direction or plane, you want to observe the velocity.

As in Stereo-Only there is a sense of scale, it possible to get the velocity by manually calculating it  based on the camera poses from frame to frame, I guess. 

It is important to note that when using the IMU, the Y-axis of Vwb2 is aligned with the gravity, which may not be the case for the Stereo only Mode.

nindanaoto, 2022-09-04:
I encountered the same problem, so I wrote my ROS wrapper to publish the velocity by integrating IMU for Stereo_Inertial mode.
https://github.com/nindanaoto/ORB_SLAM3/blob/9736d1213f1d2c1401c0115ecb3843ca6f47b327/Examples/ROS/ORB_SLAM3/src/ros_stereo_inertial.cc#L381-L401
https://qiita.com/nindanaoto/items/20858eca08aad90b5bab